COVESA/vsomeip #117: Show subscribed Clients on Service side?

kaip-e, 2020-03-10:
Hey there,
I got two Problems:
a) Is there any way to get a list of subscribed clients on service-side?

b) I tried to reproduce a client-service (player-controller) communication where the player checks the aviability of the controller via on_availability.

However i get the following issues:
```
Controller [1111.2222] NOT available.
2020-03-10 04:30:07.043687 [error] Routing info for remote service could not be found! (0124): [1111.2222.3333] 0001
```

My client and service are written based on the "vsomeip on 10minutes"-tutorial.

Also i set up some .json-files.

**Player**(client):
```
{
    "unicast" : "10.103.9.132",
    "netmask" : "255.255.0.0",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "Player",
            "id" : "0x0124"
        }
    ],
    "routing" : "Player",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.0.0.251",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "5",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```
**Controller**(service):
```
{
    "unicast" : "10.103.2.2",
    "netmask" : "255.255.0.0",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "Controller",
            "id" : "0x0123"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1111",
            "instance" : "0x2222",
            "unreliable" : "30509"
        }
    ],
    "routing" : "Controller",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "244.0.0.251",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "5",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "5",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

The .cpp looks like so:
**player.cpp**:
```
#include <iomanip>
#include <iostream>
#include <sstream>

#include <condition_variable>
#include <thread>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1111
#define SAMPLE_INSTANCE_ID 0x2222
#define SAMPLE_METHOD_ID 0x3333

std::shared_ptr <vsomeip::application> player;
std::mutex mutex;
std::condition_variable condition;

void run() {
  std::unique_lock <std::mutex> its_lock(mutex);
  condition.wait (its_lock);

  std::shared_ptr <vsomeip::message> request;
  request = vsomeip::runtime::get() -> create_request();
  request -> set_service(SAMPLE_SERVICE_ID);
  request -> set_instance(SAMPLE_INSTANCE_ID);
  request -> set_method(SAMPLE_METHOD_ID);

  std::shared_ptr <vsomeip::payload> player_payload = vsomeip::runtime::get() -> create_payload();
  std::vector< vsomeip::byte_t> player_payload_data;

  for( vsomeip::byte_t i = 0; i < 10; i++){
    player_payload_data.push_back (i % 256);
  }

  player_payload -> set_data(player_payload_data);
  request -> set_payload(player_payload);
  player -> send(request, true);
}

void on_message (const std::shared_ptr <vsomeip::message> & _response){
  std::shared_ptr <vsomeip::payload> message_payload = _response -> get_payload();
  vsomeip::length_t message_length = message_payload -> get_length();

  std::stringstream stream;
  for (vsomeip::length_t i=0; i < message_length; i++){
    stream << std::hex << (int)*(message_payload -> get_data() +i);
  }

  std::cout << "Player recieved message :" << stream.str() << std::endl;
}

void on_availability(vsomeip::service_t available_service, vsomeip::instance_t available_instance, bool is_available){
  std::cout << "Controller [" << std::setw(4) << std::setfill('0') << std::hex
            << available_service << "." << available_instance << "] " << (is_available ? "available." : "NOT available.") << std::endl;
  condition.notify_one();
}

int main(){
  player = vsomeip::runtime::get() -> create_application("Player");
  player -> init();
  player -> register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
  player -> request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
  player -> register_message_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message);

  std::thread sender (run);

  player -> start();
}
```

**controller.cpp**:
```
#include <iomanip>
#include <iostream>
#include <sstream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1111
#define SAMPLE_INSTANCE_ID 0x2222
#define SAMPLE_METHOD_ID 0x3333

std::shared_ptr <vsomeip::application> controller;

void on_message (const std::shared_ptr <vsomeip::message> & _request){
  std::shared_ptr <vsomeip::payload> message_payload = _request -> get_payload();
  vsomeip::length_t message_length = message_payload -> get_length();

  std::stringstream stream;
  for (vsomeip::length_t i = 0; i < message_length; i++){
    stream << std::hex << (int)*(message_payload -> get_data() +i );
  }

  std::cout << "Player [" << std::setw(4) << std::setfill('0') << std::hex << _request->get_client() <<"] send message: "
            << stream.str() << std::endl;

  // Response to Player baesd on msg:
  std::shared_ptr <vsomeip::message> controller_response = vsomeip::runtime::get() -> create_response (_request);
  controller_payload = vsomeip::runtime::get() -> create_payload();

  std::vector <vsomeip::byte_t> controller_payload_data;

  // response = 9 8 7 6 ... 0
  for (int i = 9; i >= 0; i--){
    // pushes i to message_payload_data as byte
    controller_payload_data.push_back (i % 256);
  }

  controller_payload -> set_data (controller_payload_data);
  controller_response -> set_payload (controller_payload);

  controller -> send (controller_response, true);
}

int main(){
  controller = vsomeip::runtime::get() -> create_application ("Controller");
  controller -> init();
  controller -> register_message_handler (SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message);
  controller -> offer_service (SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
  controller -> start();
}

```

Thank you for helping.

ManuelKugelmann, 2020-03-12:
If you're on Windows, you have to enable Multicast (e.g. via Powershell  `Set-NetIpv4Protocol -MulticastForwarding Enabled `) for the Service Discovery.

Maybe first try it locally (just add the client app entry to your server .json and run both apps on the same machine) ...

kaip-e, 2020-03-12:
@ManuelKugelmann 
I am working with two Ubuntu 14.04 machines (should have mentined that already [: )
I already enabled Mutlicast on both machines on port 244.0.0.251 . 
Also i wrote a subscribing client to a service. However that did work with no errors at all.

Also double checked the IPs of the Machines.

I jsut noticed another error on client side:
`2020-03-12 11:57:36.344119 [warning] Routing Manager seems to be inactive. Taking over...`

Guess he is using another routing file with false information even with the custom bash:
```
#!/usr/bin/bash
export VSOMEIP_CONFIGURATION=../player.json
export VSOMEIP_APPLICATION_NAME=Player
./player
```

kaip-e, 2020-03-12:
> @ManuelKugelmann
> I am working with two Ubuntu 14.04 machines (should have mentined that already [: )
> I already enabled Mutlicast on both machines on port 244.0.0.251 .
> Also i wrote a subscribing [...]

Fixed the issue:
company network did a restart while all employees where out taking a break. All dynamic IPs got repositioned

kaip-e, 2020-03-13:
So there is one question left:
Is there any way to get a list of subscribed clients on service-side?

ManuelKugelmann, 2020-03-17:
There is a subscription event you can use to populate a list. There is also a internal list of clients somewhere, but it might be at the routing manager level - have a look at `application.h` to see if the API exposes what you need.

lutzbichler, 2020-03-24:
You need to create the list within the application. To do so, use "application::register_[async_]subscription_handler" and collect the client information you receive.

closed 2020-03-24 by lutzbichler