COVESA/vsomeip #118: Is Vsomeip capable of reassembly UDP fragmented packages?

cjs102, 2020-03-11:
Hello, i have studied very closely the code of vsomeip and i have found some functions, classes, etc (tp, tp_message, and tp_reassembler) that basically includes functions to split and reconstruct messages. Right now there is no documention or examples (other than the tests) about how to do this. 

Will vsomeip be capable of such thing in the near future? If so, when?

Thanks in advance.

ManuelKugelmann, 2020-03-12:
Look into the user guide [https://github.com/GENIVI/vsomeip/blob/master/documentation/vsomeipUserGuide](url) for the necessary configuration for the SOME/IP-TP feature.
If the tests work, it should be ready for use...

lutzbichler, 2020-03-24:
SOME/IP-TP is implemented in v3 and its usage (as was pointed already pointed out by @ManuelKugelmann) is documented in the user guide. There no examples other than the tests available.

closed 2020-03-24 by lutzbichler