COVESA/vsomeip #124: Netlink send error

Emilyljj, 2020-04-30:
Hello:
When I running the helloworld example, the service program got a warning information:"Netlink send error : Invalid argument" and the Network interface is "n/a". could you please give me some recommendations on how to find out the problem.
Log are below:
2020-04-30 15:29:42.117444 [info] Parsed vsomeip configuration in 0ms
2020-04-30 15:29:42.117819 [info] Configuration module loaded.
2020-04-30 15:29:42.118327 [info] Initializing vsomeip application "".
2020-04-30 15:29:42.119654 [info] Instantiating routing manager [Host].
2020-04-30 15:29:42.119902 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-04-30 15:29:42.120563 [info] Service Discovery enabled. Trying to load module.
2020-04-30 15:29:42.123970 [info] Service Discovery module loaded.
2020-04-30 15:29:42.124141 [info] Application(unnamed, 0100) is initialized (11, 100).
2020-04-30 15:29:42.124503 [info] Starting vsomeip application "" (0100) using 2 threads I/O nice 255
2020-04-30 15:29:42.125191 [info] main dispatch thread id from application: 0100 () is: 7fb3f4630700 TID: 6007
2020-04-30 15:29:42.125242 [info] shutdown thread id from application: 0100 () is: 7fb3f3e20700 TID: 6008
2020-04-30 15:29:42.125357 [info] Watchdog is disabled!
2020-04-30 15:29:42.126284 [info] OFFER(0100): [1111.2222:0.0] (true)
2020-04-30 15:29:42.126357 [info] io thread id from application: 0100 () is: 7fb3f82c0d40 TID: 6005
2020-04-30 15:29:42.126389 [info] io thread id from application: 0100 () is: 7fb3f2e00700 TID: 6010
2020-04-30 15:29:42.127342 [warning] **Netlink send error : Invalid argument**
2020-04-30 15:29:42.127526 [info] Listening at /tmp/vsomeip-100
2020-04-30 15:29:42.127648 [info] vSomeIP 3.1.7.1 | (default)
2020-04-30 15:29:42.127914 [info] Network interface **"n/a"** state changed: up
2020-04-30 15:29:42.128866 [info] Route "n/a" state changed: up
2020-04-30 15:29:42.145365 [info] udp_server_endpoint_impl: SO_RCVBUF is: 163840
2020-04-30 15:29:42.145719 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 163840
2020-04-30 15:29:42.146809 [error] udp_server_endpoint_impl::join:set_option: Invalid argument
2020-04-30 15:29:42.149148 [info] Port configuration missing for [1111.2222]. Service is internal.
2020-04-30 15:29:42.149920 [info] SOME/IP routing ready.
^C2020-04-30 15:29:48.957438 [info] STOP OFFER(0100): [1111.2222:0.0] (true)
2020-04-30 15:29:48.964747 [info] Stopping vsomeip application "" (0100).

And my configuration is list below, I only change the unicast.
{
    "unicast":"172.26.20.29",
    "logging":
    {
        "level":"debug",
        "console":"true"
    },

    "applications":
    [
        {
            "name":"hello_world_service",
            "id":"0x4444"
        },

        {
            "name":"hello_world_client",
            "id":"0x5555"
        }
    ],

    "services":
    [
        {
            "service":"0x1111",
            "instance":"0x2222",
            "unreliable":"30509"
        },
		{
            "service":"0x1112",
            "instance":"0x2222",
            "unreliable":"30510"
        }
    ],

    "routing":"hello_world_service",
    "service-discovery":
    {
        "enable":"false"
    }
}

closed 2020-05-13 by Emilyljj

hmluqman, 2021-07-06:
@Emilyljj what is the solution of your problem?
I am facing the same issue.

AndyYangjd, 2022-06-22:
Can anybody post the solution of this issue?

Kiruthikap7, 2023-11-03:
@Emilyljj @hmluqman  @AndyYangjd  Any solution identified ?