COVESA/vsomeip #125: How to build vsomeip3 using Android Studio externalNativeBuild or in Windows system?

VishalGurav1992, 2020-05-04:


nkh-lab, 2020-05-13:
In my case I had to update CMakeLists.txt (for example for Boost stuff, due find_package( Boost) does not work for my project also if you use boost 1.7 pay attention to my pull request https://github.com/GENIVI/vsomeip/pull/127) and after that it seems that now I have possibility to communicate between services from native code.

zeerd, 2020-05-19:
Hello!

Your libraries working fine with the ndk?
I build them with ndk and cmake , but seems the dlopen failed when loading libvsomeip3-cfg.so

nkh-lab, 2020-05-19:
> Your libraries working fine with the ndk?
It works, I have an issue only if I use vsomeip with commonapi in NDK (issue about dynamic cast), which now is unresolved and works with some workaround only.

zeerd, 2020-05-22:
Me too. I put all the CPPs into only one single .so file , and the dynamic cast works.

closed 2020-11-30 by lutzbichler