COVESA/vsomeip #134: The problem of notification

Shangwei1024, 2020-05-20:
I encountered a problem:
I used /exmaples/notify-sample.cpp to notify only one service with a period of 5 seconds; using /exmaples/subscribe-sample.cpp received a message with a period of 5 seconds;
But I used /exmaples/notify-sample.cpp to notify two services, one cycle is 5 seconds, another cycle is 1 second; the message received using /exmaples/subscribe-sample.cpp is one cycle is 10 seconds, The another cycle is 1 second; I don’t know where the problem is.
The version of the vsomeip is 3.1.7;
the /exmaples/notify-sample.cpp :

#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
#include <csignal>
#endif
#include <chrono>
#include <condition_variable>
#include <iomanip>
#include <iostream>
#include <sstream>
#include <thread>
#include <mutex>

#include <vsomeip/vsomeip.hpp>

#include "sample-ids.hpp"

class service_sample {
public:
    service_sample(bool _use_tcp, uint32_t _cycle) :
            app_(vsomeip::runtime::get()->create_application()),
            is_registered_(false),
            use_tcp_(_use_tcp),
            cycle_(_cycle),
            blocked_(false),
            running_(true),
            is_offered_(false),
            offer_thread_(std::bind(&service_sample::run, this)),
            notify_thread_(std::bind(&service_sample::notify, this)) {
    }

    bool init() {
        std::lock_guard<std::mutex> its_lock(mutex_);

        if (!app_->init()) {
            std::cerr << "Couldn't initialize application" << std::endl;
            return false;
        }
        app_->register_state_handler(
                std::bind(&service_sample::on_state, this,
                        std::placeholders::_1));
        std::set<vsomeip::eventgroup_t> its_group1;
        its_group1.insert(0x1108);
        app_->offer_event(
                0x1008,
                0x0001,
                0x9002,
                its_group1,
                vsomeip::event_type_e::ET_SELECTIVE_EVENT,std::chrono::milliseconds(1000),
                false, false, nullptr, vsomeip::reliability_type_e::RT_UNKNOWN);
        // app_->offer_event(
        //         0x1008,
        //         0x0001,
        //         0x9003,
        //         its_group1,
        //         vsomeip::event_type_e::ET_SELECTIVE_EVENT,std::chrono::milliseconds(1000),
        //         false, false, nullptr, vsomeip::reliability_type_e::RT_UNKNOWN);
        
        std::set<vsomeip::eventgroup_t> its_group2;
        its_group2.insert(0x1109);
        app_->offer_event(
                0x1009,
                0x0001,
                0x8001,
                its_group2,
                vsomeip::event_type_e::ET_SELECTIVE_EVENT,std::chrono::milliseconds(5000),
                false, false, nullptr, vsomeip::reliability_type_e::RT_UNKNOWN);
        {
            std::lock_guard<std::mutex> its_lock(payload_mutex_);
            payload_ = vsomeip::runtime::get()->create_payload();
        }

        blocked_ = true;
        condition_.notify_one();
        return true;
    }

    void start() {
        app_->start();
    }

#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
    /*
     * Handle signal to shutdown
     */
    void stop() {
        running_ = false;
        blocked_ = true;
        condition_.notify_one();
        notify_condition_.notify_one();
        app_->clear_all_handler();
        stop_offer();
        offer_thread_.join();
        notify_thread_.join();
        app_->stop();
    }
#endif

    void offer() {
        std::lock_guard<std::mutex> its_lock(notify_mutex_);
        app_->offer_service(0x1009, 0x0001);
        app_->offer_service(0x1008, 0x0001);
        is_offered_ = true;
        notify_condition_.notify_one();
    }

    void stop_offer() {
        app_->stop_offer_service(0x1009, 0x0001);
        is_offered_ = false;
    }

    void on_state(vsomeip::state_type_e _state) {
        std::cout << "Application " << app_->get_name() << " is "
        << (_state == vsomeip::state_type_e::ST_REGISTERED ?
                "registered." : "deregistered.") << std::endl;

        if (_state == vsomeip::state_type_e::ST_REGISTERED) {
            if (!is_registered_) {
                is_registered_ = true;
            }
        } else {
            is_registered_ = false;
        }
    }

    void run() {
        std::unique_lock<std::mutex> its_lock(mutex_);
        while (!blocked_)
            condition_.wait(its_lock);

        bool is_offer(true);

        offer();
    }

    void notify() {
        std::shared_ptr<vsomeip::message> its_message
            = vsomeip::runtime::get()->create_request(use_tcp_);

        its_message->set_service(SAMPLE_SERVICE_ID);
        its_message->set_instance(SAMPLE_INSTANCE_ID);
        its_message->set_method(SAMPLE_SET_METHOD_ID);

        vsomeip::byte_t its_data[10];
        uint32_t its_size = 1;

        while (running_) {
            std::unique_lock<std::mutex> its_lock(notify_mutex_);
            while (!is_offered_ && running_)
                notify_condition_.wait(its_lock);
            while (is_offered_ && running_) {
                if (its_size == sizeof(its_data))
                    its_size = 1;

                for (uint32_t i = 0; i < its_size; ++i)
                    its_data[i] = static_cast<uint8_t>(i);

                {
                    std::lock_guard<std::mutex> its_lock(payload_mutex_);
                    payload_->set_data(its_data, its_size);

                    std::cout << "Setting event (Length=" << std::dec << its_size << ")." << std::endl;
                    app_->notify(0x1009, 0x0001, 0x8001, payload_);
                    app_->notify(0x1008, 0x0001, 0x9002, payload_);
                    // app_->notify(0x1008, 0x0001, 0x9003, payload_);
                }

                its_size++;

                std::this_thread::sleep_for(std::chrono::milliseconds(cycle_));
            }
        }
    }

private:
    std::shared_ptr<vsomeip::application> app_;
    bool is_registered_;
    bool use_tcp_;
    uint32_t cycle_;

    std::mutex mutex_;
    std::condition_variable condition_;
    bool blocked_;
    bool running_;

    std::mutex notify_mutex_;
    std::condition_variable notify_condition_;
    bool is_offered_;

    std::mutex payload_mutex_;
    std::shared_ptr<vsomeip::payload> payload_;

    // blocked_ / is_offered_ must be initialized before starting the threads!
    std::thread offer_thread_;
    std::thread notify_thread_;
};

#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
    service_sample *its_sample_ptr(nullptr);
    void handle_signal(int _signal) {
        if (its_sample_ptr != nullptr &&
                (_signal == SIGINT || _signal == SIGTERM))
            its_sample_ptr->stop();
    }
#endif

int main(int argc, char **argv) {
    bool use_tcp = false;
    uint32_t cycle = 5000; // default 1s

    std::string tcp_enable("--tcp");
    std::string udp_enable("--udp");
    std::string cycle_arg("--cycle");

    for (int i = 1; i < argc; i++) {
        if (tcp_enable == argv[i]) {
            use_tcp = true;
            break;
        }
        if (udp_enable == argv[i]) {
            use_tcp = false;
            break;
        }

        if (cycle_arg == argv[i] && i + 1 < argc) {
            i++;
            std::stringstream converter;
            converter << argv[i];
            converter >> cycle;
        }
    }

    service_sample its_sample(use_tcp, cycle);
#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
    its_sample_ptr = &its_sample;
    signal(SIGINT, handle_signal);
    signal(SIGTERM, handle_signal);
#endif
    if (its_sample.init()) {
        its_sample.start();
        return 0;
    } else {
        return 1;
    }
}

Shangwei1024, 2020-05-21:
I recompiled the protocol stack and the problem was solved, but I still do n’t know where the previous problem was

Shangwei1024, 2020-05-22:
I find another problem:
when i run the subscribe first, and run the notify second; the cycle is normal.but when i run the notify first, and run the subscribe second, the one cycle is 1 seconds, The another cycle is 10 second;the notify and subscribe is running in different devices.
the subscribe.cpp:
#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
#include <csignal>
#endif
#include <chrono>
#include <condition_variable>
#include <iomanip>
#include <iostream>
#include <sstream>
#include <thread>

#include <vsomeip/vsomeip.hpp>

#include "sample-ids.hpp"

// tbox "239.192.255.251",
//env VSOMEIP_CONFIGURATION=../config/vsomeip-local-subscribe.json VSOMEIP_APPLICATION_NAME=client-sample ./subscribe
//g++ subscribe-sample.cpp -o subscribe -std=c++11 -I/usr/local/include/vsomeip/ -L/usr/local/lib/ -lpthread -lvsomeip3-sd -lvsomeip3 -lvsomeip3-cfg

class client_sample {
public:
    client_sample(bool _use_tcp) :
            app_(vsomeip::runtime::get()->create_application()), use_tcp_(
                    _use_tcp) {
    }

    bool init() {
        if (!app_->init()) {
            std::cerr << "Couldn't initialize application" << std::endl;
            return false;
        }
        std::shared_ptr< vsomeip::payload > its_payload = vsomeip::runtime::get()->create_payload();
        std::vector< vsomeip::byte_t > its_payload_data;
        for (std::size_t i = 0; i < 10; ++i) its_payload_data.push_back(i % 256);
        its_payload->set_data(its_payload_data);
        std::cout << "Client settings [protocol="
                << (use_tcp_ ? "TCP" : "UDP")
                << "]"
                << std::endl;

        app_->register_state_handler(
                std::bind(&client_sample::on_state, this,
                        std::placeholders::_1));

        app_->register_message_handler(
                vsomeip::ANY_SERVICE, vsomeip::ANY_INSTANCE, vsomeip::ANY_METHOD,
                std::bind(&client_sample::on_message, this,
                        std::placeholders::_1));

        app_->register_availability_handler(vsomeip::ANY_SERVICE, vsomeip::ANY_INSTANCE,
                std::bind(&client_sample::on_availability,
                          this,
                          std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));

        // std::set<vsomeip::eventgroup_t> its_groups;
        // its_groups.insert(SAMPLE_EVENTGROUP_ID);
        // app_->request_event(
        //         SAMPLE_SERVICE_ID,
        //         SAMPLE_INSTANCE_ID,
        //         SAMPLE_EVENT_ID,
        //         its_groups,
        //         true);
        // app_->subscribe(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENTGROUP_ID);
        // std::cout<<"subscribe event..."<<std::endl;

        return true;
    }

[... 188 lines cut ...]
                (_signal == SIGINT || _signal == SIGTERM))
            its_sample_ptr->stop();
    }
#endif

int main(int argc, char **argv) {
    bool use_tcp = false;
    if(argc>1)
        std::cout<<"..."<<std::endl;
    std::string tcp_enable("--tcp");
    std::string udp_enable("--udp");

    client_sample its_sample(use_tcp);
#ifndef VSOMEIP_ENABLE_SIGNAL_HANDLING
    its_sample_ptr = &its_sample;
    signal(SIGINT, handle_signal);
    signal(SIGTERM, handle_signal);
#endif
    if (its_sample.init()) {
        its_sample.start();
        return 0;
    } else {
        return 1;
    }
}

lutzbichler, 2020-08-11:
Could you please attach your modified sample code?

closed 2020-11-10 by lutzbichler