COVESA/vsomeip #135: vsomeip v3.1.7.1 & v2.14.16 compile error

chenchangsheng-git, 2020-05-21:
compile vsomeip 2.14.16 & 3.1.7.1：
/home/share/VsomeIP/vsomeip-2.14.16/implementation/routing/src/../include/routing_manager_base.hpp:218:10: error: 'void vsomeip::routing_manager_base::erase_incoming_subscription_state(vsomeip::client_t, vsomeip::service_t, vsomeip::instance_t, vsomeip::eventgroup_t, vsomeip::event_t)' is protected
     void erase_incoming_subscription_state(client_t _client, service_t _service, instance_t _instance,
          ^
/home/share/VsomeIP/vsomeip-2.14.16/implementation/routing/src/routing_manager_impl.cpp:487:40: error: within this context
                     _eventgroup, _event);

how to solve it?

Shangwei1024, 2020-05-21:
Try to compile under root.

chenchangsheng-git, 2020-05-22:
> Try to compile under root.

I attempt to compile under root ,but also  can't solve the problem.

the error means public derived class used a protected function in basic class.

i don't know how it happen ,and how to solve it ,does anyone else have the same problem?

Shangwei1024, 2020-05-22:
I guess there is something wrong with your compilation environment. Can I take a look at the logs after your cmake？I did not encounter this problem when I compile vsomeip 2.14.16 & 3.1.7.1.

chenchangsheng-git, 2020-05-22:
-- The C compiler identification is GNU 4.9.3
-- The CXX compiler identification is GNU 4.9.3
-- Check for working C compiler: /home/share/VsomeIP/vsomeip-2.14.16/Tools/gcc/gcc-glibc-locale-internal-armv7a-vfp-neon-oe-linux-gnueabi-4.9.3/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc
-- Check for working C compiler: /home/share/VsomeIP/vsomeip-2.14.16/Tools/gcc/gcc-glibc-locale-internal-armv7a-vfp-neon-oe-linux-gnueabi-4.9.3/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/share/VsomeIP/vsomeip-2.14.16/Tools/gcc/gcc-glibc-locale-internal-armv7a-vfp-neon-oe-linux-gnueabi-4.9.3/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++
-- Check for working CXX compiler: /home/share/VsomeIP/vsomeip-2.14.16/Tools/gcc/gcc-glibc-locale-internal-armv7a-vfp-neon-oe-linux-gnueabi-4.9.3/bin/arm-oe-linux-gnueabi/arm-oe-linux-gnueabi-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.65.0
-- Found the following Boost libraries:
--   system
--   thread
--   log
--   chrono
--   date_time
--   atomic
--   log_setup
--   filesystem
--   regex
-- Boost_LIBRARY_DIR not empty using it: /home/share/VsomeIP/vsomeip-2.14.16/build/LibBoost/lib
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- Checking for module 'automotive-dlt >= 2.11'
--   No package 'automotive-dlt' found
-- Checking for module 'libsystemd'
--   No package 'libsystemd' found
-- Systemd was not found, watchdog disabled!
Predefined unicast address: 127.0.0.1
Predefined diagnosis address: 0x00
Predefined routing application: vsomeipd
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
CMake Warning at CMakeLists.txt:335 (message):
  Doxygen is not installed.  Documentation can not be built.

CMake Warning at CMakeLists.txt:374 (message):
  asciidoc is not installed.  Readme can not be built.

-- GTEST_ROOT is not defined. For building the tests the variable
             GTEST_ROOT has to be defined. Tests can not be built.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/share/VsomeIP/vsomeip-2.14.16/build
Scanning dependencies of target vsomeip

chenchangsheng-git, 2020-05-22:
> I guess there is something wrong with your compilation environment. Can I take a look at the logs after your cmake？I did not encounter this problem when I compile vsomeip 2.14.16 & 3.1.7.1.

please help me check the compilation environment,i used cross-compilation,the same environment compile v2.6.2 is normal

Shangwei1024, 2020-05-22:
the vsomeip 2.14.16& 3.1.7.1 documentation/vsomeipUserGuide:
A C++11 enabled compiler like gcc >= 5.2 is needed.
you should install gcc-v5 and g++-v5.

chenchangsheng-git, 2020-05-22:
> A C++11 enabled compiler like gcc >= 5.2 is needed.
> you should install gcc-v5 and g++-v5.

thanks a lot，i only see the vsomeip v2.6.2 need gcc >= 4.8,it's my fault.

closed 2020-05-22 by chenchangsheng-git

WillGooo, 2020-09-08:
@chenchangsheng-git i'am learning vsomeip source code recently and try to porting to arm-linux platform. would you please share you cross-compilation configuration?