COVESA/vsomeip #137: EVENT calls : publish subscribe methodology queries

krishnaruby, 2020-06-27:
I like to understand how  publish subscribe methodology works between server and client 

[image]

Once event occured at the server, server invokes notification call(publish)  to all intended subscribers(client). Is this a blocking or non blocking call ?  I  didn't find this information in vsomeIP user guide.  If this a non blocking call is there any configuration for number of dispatch thread for each server/client ?

lutzbichler, 2020-08-10:
Messages are sent within the context of the io threads. Its number can be configured per application ("threads").

krishnaruby, 2020-08-10:
Wats does the io thread mean ? Do u mean to say that dispatch threads will be created by vsomeip application and these dispatch threads will take care of notification to the appropriate client ?

lutzbichler, 2020-08-10:
Yes, io threads are the threads that run the boost io context which handles the send and receive operations.

krishnaruby, 2020-08-10:
Thanks for the information

closed 2020-08-11 by lutzbichler