COVESA/vsomeip #138: Warning: "Didn't receive a multicast SD message for 2200ms."

cfanatic, 2020-06-30:
Hi,

I was able to successfully implement and test the [Request/Response example](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#request).

However, while parsing through the [log file of the service application](https://github.com/cfanatic/vsomeip-fuzzing/blob/master/trace/vsomeip_service_fbde06de.log), I encountered a periodic sequence of warnings as shown below. The service application was started before the client application if that is of any importance.

```
2020-05-30 18:17:31.676709 [warning] Didn't receive a multicast SD message for 2200ms.
2020-05-30 18:17:31.677699 [debug] Leaving the multicast group 224.224.224.245 from 192.168.0.3
2020-05-30 18:17:31.681625 [debug] Joining to multicast group 224.224.224.245 from 192.168.0.3
2020-05-30 18:17:31.682063 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 212992
2020-05-30 18:17:33.686101 [warning] Didn't receive a multicast SD message for 2200ms.
2020-05-30 18:17:33.686846 [debug] Leaving the multicast group 224.224.224.245 from 192.168.0.3
2020-05-30 18:17:33.690809 [debug] Joining to multicast group 224.224.224.245 from 192.168.0.3
2020-05-30 18:17:33.691200 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 212992
[...]
```

Is someone able to explain to me why I was not receiving any service discovery messages? Is this due to a configuration mistake on my side?

Thanks,
Arnd

cfanatic, 2020-07-03:
Those cyclic warnings about missing multicast SD messages are gone when you add following line in `client-example.cpp` to the main() function:
```
app->offer_service(CLIENT_SERVICE_ID, CLIENT_INSTANCE_ID);
```
The `vsomeip.json` configuration file must be modified accordingly.

I consider this as a workaround. Isn't there a better solution? I do not want to client to offer any services. The client should send some SD frames, so that the service does not complain. Is this possible to achieve?

lutzbichler, 2020-07-14:
The only way to get rid of this message is to remove it from the code. Within the networks vsomeip was designed for all clients offer services and the message indicates a problem with receiving multicast messages, which sometimes happens because of driver/hw problems.

closed 2020-08-11 by lutzbichler