COVESA/vsomeip #139: Configuration module could not be loaded!

dotrung-dev, 2020-07-01:
Hello all,

I try to run hello_world
so look like I can't load configuration file

```
 VSOMEIP_CONFIGURATION=helloworld-local.json VSOMEIP_APPLICATION_NAME=hello_world_service ./hello_world_service
Configuration module could not be loaded!
```
What's my failed ?

lutzbichler, 2020-07-14:
This usually happens when the ld path does not contain the directory that contains libvsomeip3-cfg.so

Stackleon, 2020-07-26:
Hi， 
I get simiarly error on Android platform. because of process dlopen the vsomeip-cfg.so name not corrspending with defined on internal_android.hpp.
You should be checked it on Pi3+ armhf.

dotrung-dev, 2020-08-01:
I export LD_LIBRARY_PATH =/location build someip/ to solved it!

closed 2020-08-01 by dotrung-dev