COVESA/vsomeip #142: IPv6 mcast address error: `bind multicast: Invalid argument` (v3.1.14.1) - No comms

Louisjvv, 2020-07-16:
Good Day,

I am experiencing the following error when starting my vsomeip app:

2020-06-16 08:28:43.478971 [info] Parsed vsomeip configuration in 29ms
2020-06-16 08:28:43.480050 [info] Using configuration file: "./etc/someip/someipServer.json".
2020-06-16 08:28:43.488679 [info] Configuration module loaded.
2020-06-16 08:28:43.489901 [info] Client side logging for application: someipServer is enabled
2020-06-16 08:28:43.490300 [info] Initializing vsomeip application "someipServer".
2020-06-16 08:28:43.491118 [info] Instantiating routing manager [Host].
2020-06-16 08:28:43.537216 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-06-16 08:28:43.538932 [info] Service Discovery enabled. Trying to load module.
2020-06-16 08:28:43.544571 [info] Service Discovery module loaded.
2020-06-16 08:28:43.545675 [info] vsomeip tracing not enabled. . vsomeip service discovery tracing not enabled. 
2020-06-16 08:28:43.546136 [info] Application(someipServer, 2222) is initialized (11, 100).
2020-06-16 08:28:48.655470 [info] Starting vsomeip application "someipServer" (2222) using 2 threads I/O nice 255
2020-06-16 08:28:48.656906 [info] main dispatch thread id from application: 2222 (someipServer) is: 7fc68a745700 TID: 26075
2020-06-16 08:28:48.657182 [info] shutdown thread id from application: 2222 (someipServer) is: 7fc689f44700 TID: 26076
2020-06-16 08:28:48.661010 [info] Watchdog is disabled!
2020-06-16 08:28:48.663885 [info] io thread id from application: 2222 (someipServer) is: 7fc691044440 TID: 26024
2020-06-16 08:28:48.673209 [info] vSomeIP 3.1.14.1 | (default)
2020-06-16 08:28:48.666236 [info] io thread id from application: 2222 (someipServer) is: 7fc688f42700 TID: 26078
2020-06-16 08:28:48.674866 [info] Network interface "tun_fpga.vlan" state changed: up
2020-06-16 08:28:48.678693 [info] Route "ff14::4:0/128 if: tun_fpga.vlan gw: n/a" state changed: up
2020-06-16 08:28:48.679584 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936
2020-06-16 08:28:48.680042 [debug] Joining to multicast group ff14::4:0 from fd53:7cb8:383:e::6f
**2020-06-16 08:28:48.681372 [error] udp_server_endpoint_impl::join:bind multicast: Invalid argument**
2020-06-16 08:28:48.681890 [info] SOME/IP routing ready.
2020-06-16 08:28:48.683188 [warning] Route "ff00::/8 if: tun_fpga.vlan gw: n/a" state changed: up

**someipServer.json:**

{
    "unicast": "fd53:7cb8:383:3::10",
    "logging": {
        "console": "true",
        "dlt": "true",
        "level": "debug"
    },
    "tracing": {
        "enable": "false",
        "sd_enable": "false"
    },
    "applications": [
        {
            "id": "0x2222",
            "name": "someipServer",
            "num_dispatchers": 8
        }
    ],
    "services": [
        {
            "events": [
                {
                    "event": "0x0001",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0002",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0003",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0004",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0005",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0006",
                    "is_field": "false",
                    "is_reliable": "true"
                },
                {
                    "event": "0x0007",
                    "is_field": "false",
                    "is_reliable": "true"
                }
            ],
            "eventgroups": [
                {
                    "eventgroup": "0x0001",
                    "events": [
                        "0x0001"
                    ]
                },
                {
                    "eventgroup": "0x0002",
                    "events": [
                        "0x0002"
                    ]
                },
                {
                    "eventgroup": "0x0003",
                    "events": [
                        "0x0003"
                    ]
                },
                {
                    "eventgroup": "0x0004",
                    "events": [
                        "0x0004"
                    ]
                },
                {
                    "eventgroup": "0x0005",
                    "events": [
                        "0x0005"
                    ]
                },
                {
                    "eventgroup": "0x0006",
                    "events": [
                        "0x0006"
                    ]
                },
                {
                    "eventgroup": "0x0007",
                    "events": [
                        "0x0007"
                    ]
                }
            ],
            "instance": "0x0001",
            "service": "0x6fe7",
            "reliable": "29180"
        }
    ],
    "routing": "someipServer",
    "service-discovery": {
        "cyclic_offer_delay": "1000",
        "enable": "true",
        "initial_delay_max": "100",
        "initial_delay_min": "10",
        "multicast": "ff14::4:0",
        "port": "30490",
        "protocol": "udp",
        "repetitions_base_delay": "200",
        "repetitions_max": "3",
        "request_response_delay": "1500",
        "ttl": "3"
    }
}

Louisjvv, 2020-07-28:
Here is a patch that solved this issue:

```
diff --git a/implementation/endpoints/src/udp_server_endpoint_impl.cpp b/implementation/endpoints/src/udp_server_endpoint_impl.cpp
index 9d22d51..ce1bb2d 100644
--- a/implementation/endpoints/src/udp_server_endpoint_impl.cpp
+++ b/implementation/endpoints/src/udp_server_endpoint_impl.cpp
@@ -303,12 +303,26 @@ void udp_server_endpoint_impl::join(const std::string &_address) {
 
             boost::system::error_code ec;
 
+            bool is_v4(false);
+            bool is_v6(false);
+            {
+                std::lock_guard<std::mutex> its_lock(local_mutex_);
+                is_v4 = local_.address().is_v4();
+                is_v6 = local_.address().is_v6();
+            }
+
             if (multicast_recv_buffer_.empty())
                 multicast_recv_buffer_.resize(VSOMEIP_MAX_UDP_MESSAGE_SIZE, 0);
 
             if (!multicast_local_) {
-                multicast_local_ = std::unique_ptr<endpoint_type>(
-                    new endpoint_type(boost::asio::ip::address_v4::any(), local_port_));
+                if (is_v4) {
+                    multicast_local_ = std::unique_ptr<endpoint_type>(
+                        new endpoint_type(boost::asio::ip::address_v4::any(), local_port_));
+                }
+                if (is_v6) {
+                    multicast_local_ = std::unique_ptr<endpoint_type>(
+                        new endpoint_type(boost::asio::ip::address_v6::any(), local_port_));
+                }
             }
 
             if (!multicast_socket_) {
@@ -360,14 +374,6 @@ void udp_server_endpoint_impl::join(const std::string &_address) {
                 receive_multicast(multicast_id_);
             }
 
-            bool is_v4(false);
-            bool is_v6(false);
-            {
-                std::lock_guard<std::mutex> its_lock(local_mutex_);
-                is_v4 = local_.address().is_v4();
-                is_v6 = local_.address().is_v6();
-            }
-
             if (is_v4) {
                 multicast_socket_->set_option(ip::udp_ext::socket::reuse_address(true));
                 multicast_socket_->set_option(
```

lutzbichler, 2020-08-11:
Merged to v3.1.16.1

closed 2020-08-11 by lutzbichler