COVESA/vsomeip #143: there are some compile issues on Android platform

Stackleon, 2020-07-24:
Hello

I try integration vsomeip library to Android platform. but I catch some compile issue. And I has fixed it.  please to review.

issue 1:
compile the vsomeip.so has issue

vendor/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:648: error: undefined reference to 'vsomeip_v3::logger::message::message(vsomeip_v3::logger::level_e)'
vendor/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:648: error: undefined reference to 'vsomeip_v3::logger::message::~message()'
vendor/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:648: error: undefined reference to 'vsomeip_v3::logger::message::~message()'

fix it should add logger cpp file. like this:

libvsomeip_srcs = [
   "implementation/endpoints/**/*.cpp",
   "implementation/logging/**/*.cpp",
   "implementation/tracing/**/*.cpp",
   "implementation/message/**/*.cpp",
   "implementation/routing/**/*.cpp",
   "implementation/runtime/**/*.cpp",
   "implementation/utility/**/*.cpp",
   "implementation/plugin/**/*.cpp",
   "implementation/security/**/*.cpp",
   "implementation/logger/**/*.cpp",
]

issue 2:
It's miss some header files that on compile the vsomeip-sd.so

vsomeip/implementation/service_discovery/src/../include/ipv4_option_impl.hpp:9:10: fatal error: 'boost/asio/ip/address.hpp' file not found #include <boost/asio/ip/address.hpp>

vsomeip/implementation/service_discovery/src/../include/../../message/include/message_base_impl.hpp:9:10: fatal error: 'boost/thread.hpp' file not found
#include <boost/thread.hpp>

It's should be add relate to the liboost header files. there is add libboost_thread to fix it. like this:

cc_library_shared {
    name: "libvsomeip_sd",
    vendor: true,

    srcs: libvsomeip_sd_srcs,

    defaults: [
        "vsomeip_defaults"
    ],

    rtti: true,

    local_include_dirs: [
        "interface",
        "implementation/helper/1.70"
    ],

    shared_libs: [
        "libvsomeip",
        "libboost_thread",
    ]
}

issue 3:
 Hello_world example has some compile error . 
I shall not paste error log that relate to issues. I shall paste a Android.bp file.
3-1 : add paramter as vendor:true
3-1 : add share library liblog
3-3 : fix compile cpp file name: hello_world_service_main.cpp  hello_world_client_main.

cc_defaults {
    name: "vsomeip_hello_world_defaults",
    vendor: true,

    cppflags: [
        "-std=c++11",
        "-Wno-unused-parameter",
    ],

    shared_libs: [
        "libvsomeip",
        "libvsomeip_cfg",
        "libvsomeip_e2e",
        "libvsomeip_sd",
        "liblog",
    ],
}

cc_binary {
    name: "vsomeip_hello_world_service",
    vendor: true,
    defaults: ["vsomeip_hello_world_defaults"],

    srcs: [
        "hello_world_service_main.cpp",
    ],
}

cc_binary {
    name: "vsomeip_hello_world_client",
    vendor: true,
    defaults: ["vsomeip_hello_world_defaults"],

    srcs: [
        "hello_world_client_main.cpp",
    ],
}

prebuilt_etc {
    name: "helloworld-local.json",
    sub_dir: "vsomeip",
    src: "helloworld-local.json",
}

Hope it do better on Android plateform.

sincere Leo

Stackleon, 2020-07-24:
There are have error on runtime.
the process can not load libvsomeip3-cfg.so.  should change library name on the Android.bp that to ensure correspond with defined on the internal_android.hpp
```
#define VSOMEIP_CFG_LIBRARY                     "libvsomeip3-cfg.so"

#define VSOMEIP_SD_LIBRARY                      "libvsomeip3-sd.so"

#define VSOMEIP_E2E_LIBRARY                     "libvsomeip3-e2e.so"
```

nkh-lab, 2020-07-25:
Hi Leo
I've created corresponding pull-request which fixes your remarks and does some other improvements according to AOSP usage:
https://github.com/GENIVI/vsomeip/pull/144

BR,
Nikolay

closed 2020-07-25 by Stackleon