COVESA/vsomeip #147: Compilation of tests is failing

shubhu1006, 2020-08-05:
Hi,
I am trying to compile tests in vsomeip, but it is facing few issues:-
1. Required google test framework **version 1.7.0** is not available, link specified in user guide is broken. So i downloaded **version 1.10.x**. Will it work?
2. When i try to compile using version 1.10 it is not able to find the files :- 
[image]
3. I think issue is with the path, so i tried manually to edit the path but after that i faced following issue:- 
[image]

lutzbichler, 2020-08-10:
1) Yes, it does work with gtest v1.10.0
2/3) This looks as if you did not set some requires switches when configuring. Please post your cmake call.

shubhu1006, 2020-08-11:
Here is my cmakelist file. 
[CMakeLists.txt](https://github.com/GENIVI/vsomeip/files/5054848/CMakeLists.txt)

lutzbichler, 2020-08-11:
Please post how you call cmake or the full CMakeOutput.log.

shubhu1006, 2020-08-11:
cd vsomeip/test/
mkdir build
cd build
cmake .. 

[CMakeOutput.log](https://github.com/GENIVI/vsomeip/files/5055262/CMakeOutput.log)

lutzbichler, 2020-08-11:
This not the correct way to build the tests. Within <vsomeip>/build do:

> export GTEST_ROOT=<PATH_TO_GTEST>
> cmake -DTEST_IP_MASTER=<IP_MASTER> -DTEST_IP_SLAVE=<IP_SLAVE> ..
> make && make build_tests

There are additional defines for specific tests (TEST_IP_SLAVE_SECOND, TEST_UID, TEST_GID).

shubhu1006, 2020-08-11:
Hi I did the same earlier but results were same. I tried again and still same issue

lutzbichler, 2020-08-11:
Please log what you exactly do on your console and which error messages are logged.
Does your build directory have the same parent directoy than the test directory?

shubhu1006, 2020-08-11:
Hi @lutzbichler 
I have changed the build directory now, earlier i was compiling from **/test/build** now i changed i to **vsomeip/build** .
I ran the following commands as shown in pic below:- 
[image]
It seems it is not configured properly yet.
Still i tried to build and got error related to build_tests, as in makefile it is not able to find rule for build_tests (if you see in the last)
[image]

lutzbichler, 2020-08-11:
It is not happy with the directory that GTEST_ROOT points to. Therefore the Makefile for the tests is no generated.

shubhu1006, 2020-08-11:
Hi @lutzbichler ,
I dont know whats wrong with the path because echoing results in path correctly
[image]

When i put hardcode path (set (GTEST_ROOT /home/shubh/googletest) ) in Cmake file it is working correct

shubhu1006, 2020-08-11:
I found issue with first if condition

`**if (NOT GTEST_ROOT)**
    if (DEFINED ENV{GTEST_ROOT})
        set(GTEST_ROOT $ENV{GTEST_ROOT})
    else()
        set(GTEST_ROOT "n/a" CACHE STRING "Path to root folder of googletest. Must be set for building the tests.")
    endif()
endif()`

Document says :

> if(NOT <condition>)
>     True if the condition is not true.
So here <condition> = GTEST_ROOT will be true always and hence first if will result in false and it will not execute inside if condition.
After commenting this if, it is working

goncaloalmeida, 2023-02-17:
@shubhu1006 this problem still occurs?

goncaloalmeida, 2023-02-23:
Closing the ticket -> if the problem still occurs please reopen the ticket.

shubhu1006, 2023-03-05:
Hi @goncaloalmeida, I have already pointed where actually issue is. It will be great if you can confirm the same.

shubhu1006, 2023-03-05:
> I found issue with first if condition
>
> ```
> if (NOT GTEST_ROOT)
>     if (DEFINED ENV{GTEST_ROOT})
>         set(GTEST_ROOT $ENV{GTEST_ROOT})
>     else()
>         set(GTEST_ROOT "n/a" CACHE STRING "Path to [...]

Hi @goncaloalmeida, I have already pointed where actually issue is. It will be great if you can confirm the same.

joao-d, 2023-04-14:
The first if condition, if (NOT GTEST_ROOT), will always be false because GTEST_ROOT is already defined as an empty string at this point in the CMake script. 
To check whether GTEST_ROOT is not defined, it can be use the NOT DEFINED operator instead of NOT: 
if (NOT DEFINED GTEST_ROOT) 
    if (DEFINED ENV{GTEST_ROOT}) 
        set(GTEST_ROOT $ENV{GTEST_ROOT}) 
    else() 
        set(GTEST_ROOT "n/a" CACHE STRING "Path to root folder of googletest. Must be set for building the tests.") 
    endif() 
endif() 
 
This will only enter the if block if GTEST_ROOT is not defined at all, whether it is an empty string or not 
Do you want to create a PR to make this change?