COVESA/vsomeip #155: Error running first service-example

SP3coder, 2020-08-24:
I tried running service example as directed..but got the following error:
[2020-08-24 16:01:27.588882] [0x00007f1a2f6b9e40] [error]   Loading failed: (libvsomeip-cfg.so.2: cannot open shared object file: No such file or directory).

This issue was also reported earlier but even after changing LD_LIBRARY_PATH variable it is not resolved. I am using the lastest release.

gunnarx, 2020-08-24:
> This issue was also reported earlier but even after changing LD_LIBRARY_PATH variable it is not resolved. 

OK.  Did you also confirm that the file libvsomeip-cfg.so.2 exists in the directory that LD_LIBRARY_PATH points to?  Has it been installed?  

Also, please provide the ticket number or other location where it was previously reported as you said, so that someone who wants to help can see the full picture.

SP3coder, 2020-08-24:
Hi, Earlier it was reported at : https://github.com/GENIVI/vsomeip/issues/38 

I checked the files but things did no seem to work. I can explain the workflow..i guess that might help...i took the release 3.1.16.1....renamed it to vsomeip and then accessed it via Terminal at /home/vsomeip. Create two directories..just put the service code as it is. And used cmake to build and run but go the error.
[image]
[image]

Putting these images for reference.

SP3coder, 2020-08-24:
Environment variable has to be set correctly.

closed 2020-08-24 by SP3coder