COVESA/vsomeip #158: Is it possible for vsomeip2 app to communicate with vsomeip3 app

rlokanathan, 2020-09-23:
Hi, currently, I am using common API to create my client app. My common API application uses vsomeip2. Whereas my service app is using vsomeip3 without common API. May I know are there any ways to make this communication successful?

Looking forward for reply.

lutzbichler, 2020-11-10:
You must communicate using UDP/TCP and you must not use selective events. Local communication does not work. Please note that vSomeIP 3.x has a compatibility layer that enables it usage for application compiled to vSomeIP 2.x.

closed 2020-11-10 by lutzbichler

rlokanathan, 2020-11-11:
Currently, I have been using TCP/UDP. My service is running in my Linux VM and my client app is running in windows host. Yet it doesnt work. However, once I replace both sides to vsomeip2 or vsomeip3 then it works fine.

Arindom-2712, 2023-01-04:
I am also facing a similar issue. 
My client is  based on someip3 , running on Android.
Server is someip2 based running on linux.

Expecting a solution !