COVESA/vsomeip #159: copy constructor function error

xiaostang, 2020-09-28:
`serviceinfo::serviceinfo(const serviceinfo& _other) :
    group_(_other.group_),
    service_(_other.service_),
    instance_(_other.instance_),
    major_(_other.major_),
    minor_(_other.minor_),
    ttl_(_other.ttl_),
    reliable_(_other.reliable_),
    unreliable_(_other.unreliable_),
    requesters_(_other.requesters_),
    is_local_(_other.is_local_),
    is_in_mainphase_(_other.is_in_mainphase_)
    {} `
group_(_other.group_) if this is error

xiaostang, 2020-10-14:
### group_(other.group) if this is error???
[image]

lutzbichler, 2020-11-10:
Servicegroups are a leftover that are currently not used (therefore "group_" is always NULL).

closed 2020-11-10 by lutzbichler

ByteXiaoTang, 2020-11-11:
> Servicegroups are a leftover that are currently not used (therefore "group_" is always NULL).

we may try to not leave a hole for later generations