COVESA/vsomeip #162: Reboot detection triggered when client starts up first

Louisjvv, 2020-10-08:
Good Day,

I am experiencing an issue where a reboot is detected when a client application starts up before a server application.

`2020-09-06 22:35:32.460843 [info] Parsed vsomeip configuration in 1ms
2020-09-06 22:35:32.461414 [info] Using configuration file: "./tmp/someip/someip_tcp_client.json".
2020-09-06 22:35:32.461817 [info] Configuration module loaded.
2020-09-06 22:35:32.462052 [info] Initializing vsomeip application "someipTCPClient".
2020-09-06 22:35:32.462761 [info] Instantiating routing manager [Host].
2020-09-06 22:35:32.463732 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-09-06 22:35:32.464832 [info] Service Discovery enabled. Trying to load module.
2020-09-06 22:35:32.471341 [info] Service Discovery module loaded.
2020-09-06 22:35:32.472003 [info] vsomeip tracing enabled . vsomeip service discovery tracing enabled 
2020-09-06 22:35:32.472454 [info] Application(someipTCPClient, 1111) is initialized (11, 100).
2020-09-06 22:35:32.474700 [info] Starting vsomeip application "someipTCPClient" (1111) using 2 threads I/O nice 255
2020-09-06 22:35:32.478008 [info] Watchdog is disabled!
2020-09-06 22:35:32.479181 [info] io thread id from application: 1111 (someipTCPClient) is: 7f52e8758740 TID: 1710
2020-09-06 22:35:32.484417 [info] vSomeIP 3.1.16.1 | (default)
2020-09-06 22:35:32.485987 [info] Network interface "enp1s0" state changed: up
2020-09-06 22:35:32.489101 [info] Route "ff14::4:0/128 if: enp1s0 gw: n/a" state changed: up
2020-09-06 22:35:32.489895 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936
2020-09-06 22:35:32.490302 [debug] Joining to multicast group ff14::4:0 from fd53:7cb8:383:3::100
2020-09-06 22:35:32.491002 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 1703936
2020-09-06 22:35:32.491709 [info] SOME/IP routing ready.
2020-09-06 22:35:32.492918 [warning] Route "ff00::/8 if: enp1s0 gw: n/a" state changed: up
2020-09-06 22:35:32.493587 [info] io thread id from application: 1111 (someipTCPClient) is: 7f52e4671700 TID: 1719
2020-09-06 22:35:32.494689 [info] shutdown thread id from application: 1111 (someipTCPClient) is: 7f52e5673700 TID: 1717
2020-09-06 22:35:32.495335 [info] main dispatch thread id from application: 1111 (someipTCPClient) is: 7f52e5e74700 TID: 1716
2020-09-06 22:35:32.496467 [info] REQUEST(1111): [1234.0001:4.4294967295]
2020-09-06 22:35:32.498032 [info] Listening at /tmp/vsomeip-1111
2020-09-06 22:35:42.484977 [info] vSomeIP 3.1.16.1 | (default)
2020-09-06 22:35:43.709746 [info] REGISTER EVENT(1111): [1234.0001.5678:is_provider=false]
2020-09-06 22:35:43.710344 [info] SUBSCRIBE(1111): [1234.0001.5678:ffff:4]
**2020-09-06 22:35:43.755183 [info] Reboot detected: IP=fd53:7cb8:383:3::101
2020-09-06 22:35:43.756489 [info] expire_services for address: fd53:7cb8:383:3::101 : delete service/instance 1234.0001 port [0,0] reliability=false**`

Furthermore, when the server reboot is detected the client expires its subscriptions to the service and according to the log the client re-subscribes, but after this it does not receive any event notifications from the server.

`2020-09-06 22:35:43.697884 [info] REGISTER EVENT(1111): [1234.0001.5678:is_provider=false]
2020-09-06 22:35:43.698443 [info] SUBSCRIBE(1111): [1234.0001.5678:ffff:4]`

From the server's side the re-subscription can be seen:

`2020-09-06 22:35:46.797382 [info] Parsed vsomeip configuration in 1ms
2020-09-06 22:35:46.798288 [info] Using configuration file: "./tmp/someip/someip_tcp_server.json".
2020-09-06 22:35:46.798714 [info] Configuration module loaded.
2020-09-06 22:35:46.798993 [info] Initializing vsomeip application "someipTCPServer".
2020-09-06 22:35:46.799389 [info] Instantiating routing manager [Host].
2020-09-06 22:35:46.800592 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-09-06 22:35:46.801605 [info] Service Discovery enabled. Trying to load module.
2020-09-06 22:35:46.807178 [info] Service Discovery module loaded.
2020-09-06 22:35:46.807956 [info] vsomeip tracing enabled . vsomeip service discovery tracing enabled 
2020-09-06 22:35:46.808391 [info] Application(someipTCPServer, 2222) is initialized (11, 100).
2020-09-06 22:35:46.810389 [info] REGISTER EVENT(2222): [1234.0001.5678:is_provider=true]
2020-09-06 22:35:46.812748 [info] Starting vsomeip application "someipTCPServer" (2222) using 2 threads I/O nice 255
2020-09-06 22:35:46.815622 [info] Watchdog is disabled!
2020-09-06 22:35:46.816625 [info] io thread id from application: 2222 (someipTCPServer) is: 7f368a5ba740 TID: 7151
2020-09-06 22:35:46.821688 [info] vSomeIP 3.1.16.1 | (default)
2020-09-06 22:35:46.823061 [info] Network interface "enp1s0" state changed: up
2020-09-06 22:35:46.825157 [info] Route "ff14::4:0/128 if: enp1s0 gw: n/a" state changed: up
2020-09-06 22:35:46.825787 [info] udp_server_endpoint_impl: SO_RCVBUF is: 1703936
2020-09-06 22:35:46.826387 [debug] Joining to multicast group ff14::4:0 from fd53:7cb8:383:3::101
2020-09-06 22:35:46.826767 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 1703936
2020-09-06 22:35:46.827440 [info] SOME/IP routing ready.
2020-09-06 22:35:46.828390 [warning] Route "ff00::/8 if: enp1s0 gw: n/a" state changed: up
2020-09-06 22:35:46.828787 [info] io thread id from application: 2222 (someipTCPServer) is: 7f36874d5700 TID: 7158
2020-09-06 22:35:46.829971 [info] shutdown thread id from application: 2222 (someipTCPServer) is: 7f36884d7700 TID: 7156
2020-09-06 22:35:46.830941 [info] main dispatch thread id from application: 2222 (someipTCPServer) is: 7f3688cd8700 TID: 7155
2020-09-06 22:35:46.831526 [info] OFFER(2222): [1234.0001:1.0] (true)
2020-09-06 22:35:46.833844 [info] Listening at /tmp/vsomeip-2222
**2020-09-06 22:35:46.736603 [info] REMOTE SUBSCRIBE(0000): [1234.0001.5678] from fd53:7cb8:383:3::100:40737 reliable was accepted**`

I am running these applications on separate host machines. Upon an initial investigation it seems that the difficulties I am encountering originate in **service_discovery_impl::is_reboot** lines 599-602 (version 3.1.16.1):

**if (its_old_reboot_flag && _reboot_flag
                    && its_old_session >= _session) {
                result = true;
            }**

I am also attaching my configuration files for reference.

**Server**
`{
    "unicast": "fd53:7cb8:383:3::101",
    "logging": {
        "console": "true",
        "dlt": "true",
        "level": "debug"
    },
    "tracing": {
        "enable": "true",
        "sd_enable": "true"
    },
    "applications": [
        {
            "id": "0x2222",
            "name": "someipTCPServer",
            "num_dispatchers": 4
        }
    ],
    "services": [
        {
            "events": [
                {
                    "event": "0x5678",
                    "is_field": "false",
                    "is_reliable": "true"
                }
            ],
            "eventgroups": [
                {
                    "eventgroup": "0x5678",
                    "events": [
                        "0x5678"
                    ]
                }
            ],
            "instance": "0x0001",
            "name": "Service",
            "service": "0x1234",
            "reliable": {
                "port": "29180"
            }
        }
    ],
    "routing": "someipTCPServer",
    "service-discovery": {
        "enable": "true",
        "multicast": "ff14::4:0",
        "port": "30490",
        "protocol": "udp",
        "initial_delay_min": "10",
        "initial_delay_max": "100",
        "repetitions_base_delay": "200",
        "repetitions_max": "3",
        "ttl": "16777215",
        "cyclic_offer_delay": "1000",
        "request_response_delay": "1500"
    }
}`

**Client:**

`{
    "unicast": "fd53:7cb8:383:3::100",
    "logging": {
        "console": "true",
        "dlt": "true",
        "level": "debug"
    },
    "tracing": {
        "enable": "true",
        "sd_enable": "true"
    },
    "applications": [
        {
            "id": "0x1111",
            "name": "someipTCPClient",
            "num_dispatchers": 4
        }
    ],
    "clients": [
        {
            "service": "0x1234",
            "instance": "0x0001",
            "reliable": []
        }
    ],
    "routing": "someipTCPClient",
    "service-discovery": {
        "enable": "true",
        "multicast": "ff14::4:0",
        "port": "30490",
        "protocol": "udp",
        "initial_delay_min": "10",
        "initial_delay_max": "100",
        "repetitions_base_delay": "200",
        "repetitions_max": "3",
        "ttl": "16777215",
        "cyclic_offer_delay": "1000",
        "request_response_delay": "1500"
    }
}
`

Do you have any advice on what exactly the problem could be here?

lutzbichler, 2020-11-30:
Could you check whether pull request #179 "Bug-Fixes regarding IPv6 Multicast and Multicast Endpoint Clean-up" helps?

lutzbichler, 2020-12-04:
Fixed with v3.1.20.2

closed 2020-12-04 by lutzbichler

sdunnagan, 2021-12-03:
This seems to still be an issue in v3.1.20.3.

2021-11-03 12:15:00.975838 [info] Reboot detected: IP=192.168.1.63