COVESA/vsomeip #167: dynamic_cast not work on Android

bum2free, 2020-10-23:
The following part failed to work on Android:
```
//implementation/plugin/src/plugin_manager_impl.cpp
bool application_impl::init() {
...
    auto its_configuration_plugin = std::dynamic_pointer_cast<configuration_plugin>(its_plugin);
...
}
```
Is there any solution for it? Thanks a lot~
-------
The environemt of the AOSP is:
- NDK Version: R18
- Build sytem: AOSP
- Host OS: Ubuntu 18.04
- Compiler: Clang++ (9.0.3)
- TARGET_ARCH: armv8a

nkh-lab, 2020-10-26:
Hi, I had the same issue when I used CommonAPI via vsomeip and only workaround helped me to fix it. More details here: https://github.com/nkh-lab/genivi-capi-someip-examples/blob/ndk/CMakeLists.txt

bum2free, 2020-10-26:
> Hi, I had the same issue when I used CommonAPI via vsomeip and only workaround helped me to fix it. More details here: https://github.com/nkh-lab/genivi-capi-someip-examples/blob/ndk/CMakeLists.txt

Hi, thanks for the info. I am not quite catching it... Would you please elaborate more how this is realized?

# Fix: swap type info positions in linker by changing libraries loading order

nkh-lab, 2020-10-26:
We have the same typeinfo for vsomeip_v3::configuration_plugin in libvsomeip3.so and libvsomeip3-cfg.so and dynamic_pointer_cast uses the wrong one (for example first entry in typeinfo table) in our case, so we change the loading order of mentioned libraries and after that  dynamic_pointer_cast will use correct one. It is not fix but workaround.

bum2free, 2020-10-26:
> We have the same typeinfo for vsomeip_v3::configuration_plugin in libvsomeip3.so and libvsomeip3-cfg.so and dynamic_pointer_cast uses the wrong one (for example first entry in typeinfo table) in our [...]

Thank you~Normally, when the app started, libvsomeip3.so is loaded first and then libvsomeip3-cfg.so(by dlopen). Do you mean that it is adjusted that firstly load libvsomeip3-cfg.so and then libvsomeip3.so? If so, how this workaround can be implemented?

nkh-lab, 2020-10-26:
I've set order and load it explicitly in mentioned CMake file (line 30 and line 131) otherwise it would be load via dlopen in runtime (e.g. by vsomeip plugin_manager_impl::load_library() ) but in order which doesn't work in our case.

bum2free, 2020-10-26:
Very appreciate~~Now I am fully understand~~ I will have a try on aosp build, check whether this can be done in Android.bp

nkh-lab, 2020-10-26:
Strange, but under AOSP I did not faced with such an issue, here are my *.bp files: https://github.com/nkh-lab/genivi-capi-someip-examples/tree/master/aosp

bum2free, 2020-10-27:
Genuis! Thanks for the great help. Now it works.

Orignally my app only have libvsomeip3.so linked in Android.bp, (I thought others not needed as they are designed as plugins), after adding them all. problem resolved. Here is the details:

Android.bp: (orignal, dynamic_cast not work)
```
shared_libs: [
"libvsomeip3",
"liblog",
],
```

Android.bp (works now)
```
shared_libs: [
"libvsomeip3",
"libvsomeip3-cfg",
"libvsomeip3-e2e",
"libvsomeip3-sd",
"liblog",
],
```

Android.bp (segment fault, tbd)
```
shared_libs: [
"libvsomeip3",
"libvsomeip3-cfg",
"liblog",
],
```

closed 2020-10-27 by bum2free