COVESA/vsomeip #169: Client (QNX) can't communicate with Linux Service

maingig, 2020-11-05:
I am using vsomeip version 3.1.16.1.  I have a board running QNX and a my Linux box connected to a router.  I can ping each platform from the other.  I am using CommonAPI as well.  When I start up the Linux service I get this output:

My config files look exactly like those [here](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#devices) except for the ip addresses.

Port configuration missing for [ff.1]. Service is internal.

Service log:
`[CAPI][INFO] Loading configuration file '/home/fmalouf/Dev/gitlab/build-smart-antenna-services/cfg/gnss-commonapi.ini'
[CAPI][INFO] Using default binding 'someip'
[CAPI][INFO] Using default shared library folder '/home/fmalouf/Dev/gitlab/build-smart-antenna-services/lib'
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IAGnssBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IAGnssRilBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IAGnssRil:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IAGnss:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssBatchingBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssBatching:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssBroadcast:v1_1" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssConfiguration:v1_1" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssDebug:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssGeofencingBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssGeofencing:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssMeasurementBroadcast:v1_1" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssMeasurement:v1_1" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssNavigationMessageBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssNavigationMessage:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssNiBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssNi:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnss:v1_1" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssXtraBroadcast:v1_0" stub adapter.
[CAPI][INFO] Registering function for creating "com.nio.hardware.gnss.idl.IGnssXtra:v1_0" stub adapter.
[CAPI][DEBUG] Loading library for local:com.nio.hardware.gnss.idl.IGnss:v1_1:com.nio.hardware.gnss.idl.IGnss stub.
[CAPI][DEBUG] Loading interface library "libIGnssService-someip.so" succeeded.
[CAPI][INFO] Registering stub for "local:com.nio.hardware.gnss.idl.IGnss:v1_1:com.nio.hardware.gnss.idl.IGnss"
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IAGnss:v1_0:com.nio.hardware.gnss.idl.IAGnss <--> [00fb.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IAGnssBroadcast:v1_0:com.nio.hardware.gnss.idl.IAGnssBroadcast <--> [00fc.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IAGnssRil:v1_0:com.nio.hardware.gnss.idl.IAGnssRil <--> [00fd.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IAGnssRilBroadcast:v1_0:com.nio.hardware.gnss.idl.IAGnssRilBroadcast <--> [00fe.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnss:v1_1:com.nio.hardware.gnss.idl.IGnss <--> [00ff.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssBatching:v1_0:com.nio.hardware.gnss.idl.IGnssBatching <--> [0101.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssBatchingBroadcast:v1_0:com.nio.hardware.gnss.idl.IGnssBatchingBroadcast <--> [0102.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssBroadcast:v1_1:com.nio.hardware.gnss.idl.IGnssBroadcast <--> [0100.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssConfiguration:v1_1:com.nio.hardware.gnss.idl.IGnssConfiguration <--> [0104.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssDebug:v1_0:com.nio.hardware.gnss.idl.IGnssDebug <--> [0103.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssGeofencing:v1_0:com.nio.hardware.gnss.idl.IGnssGeofencing <--> [0105.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssGeofencingBroadcast:v1_0:com.nio.hardware.gnss.idl.IGnssGeofencingBroadcast <--> [0106.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssMeasurement:v1_1:com.nio.hardware.gnss.idl.IGnssMeasurement <--> [0107.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssMeasurementBroadcast:v1_1:com.nio.hardware.gnss.idl.IGnssMeasurementBroadcast <--> [0108.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssNavigationMessage:v1_0:com.nio.hardware.gnss.idl.IGnssNavigationMessage <--> [0109.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssNavigationMessageBroadcast:v1_0:com.nio.hardware.gnss.idl.IGnssNavigationMessageBroadcast <--> [010a.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssNi:v1_0:com.nio.hardware.gnss.idl.IGnssNi <--> [010b.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssNiBroadcast:v1_0:com.nio.hardware.gnss.idl.IGnssNiBroadcast <--> [010c.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssXtra:v1_0:com.nio.hardware.gnss.idl.IGnssXtra <--> [010d.0001(1.0)]
[CAPI][DEBUG] Added address mapping: local:com.nio.hardware.gnss.idl.IGnssXtraBroadcast:v1_0:com.nio.hardware.gnss.idl.IGnssXtraBroadcast <--> [010e.0001(1.0)]
2020-10-05 11:20:05.083037 [info] Parsed vsomeip configuration in 0ms
2020-10-05 11:20:05.083809 [info] Using configuration file: "/home/fmalouf/Dev/gitlab/build-smart-antenna-services/cfg/gnss-local.json".
2020-10-05 11:20:05.084086 [info] Configuration module loaded.
2020-10-05 11:20:05.084251 [info] Initializing vsomeip application "gnss-service".
2020-10-05 11:20:05.084552 [info] Instantiating routing manager [Host].
2020-10-05 11:20:05.085453 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-10-05 11:20:05.086203 [info] Service Discovery enabled. Trying to load module.
2020-10-05 11:20:05.090818 [info] Service Discovery module loaded.
2020-10-05 11:20:05.091425 [info] Application(gnss-service, 1277) is initialized (11, 100).
2020-10-05 11:20:05.096808 [info] Starting vsomeip application "gnss-service" (1277) using 2 threads I/O nice 255
2020-10-05 11:20:05.099566 [info] Watchdog is disabled!
2020-10-05 11:20:05.100546 [info] io thread id from application: 1277 (gnss-service) is: 7f66b661c700 TID: 3086
2020-10-05 11:20:05.100936 [info] main dispatch thread id from application: 1277 (gnss-service) is: 7f66b5e1b700 TID: 3087
2020-10-05 11:20:05.100680 [info] io thread id from application: 1277 (gnss-service) is: 7f66affff700 TID: 3090
2020-10-05 11:20:05.101287 [info] shutdown thread id from application: 1277 (gnss-service) is: 7f66b561a700 TID: 3088
2020-10-05 11:20:05.102559 [info] vSomeIP 3.1.16.1 | (default)
2020-10-05 11:20:05.103489 [info] Network interface "eno1" state changed: up
[... 63 lines cut ...]
1970-00-01 04:24:37.779250 [info] REGISTER EVENT(0022): [00ff.0001.a4db:is_provider=false]
1970-00-01 04:24:37.779250 [debug] Joining to multicast group 224.224.224.245 from 172.20.1.6
1970-00-01 04:24:37.779250 [info] REGISTER EVENT(0022): [00ff.0001.a4dc:is_provider=false]
1970-00-01 04:24:37.779260 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 1703936
1970-00-01 04:24:37.779260 [info] REGISTER EVENT(0022): [00ff.0001.a4dd:is_provider=false]
1970-00-01 04:24:37.779260 [info] REGISTER EVENT(0022): [00ff.0001.a4de:is_provider=false]
[... cut ...]
1970-00-01 04:24:37.779270 [info] REGISTER EVENT(0022): [00ff.0001.a4e4:is_provider=false]
1970-00-01 04:24:37.779270 [info] REQUEST(0022): [00ff.0001:1.4294967295]
1970-00-01 04:24:37.779280 [info] Listening at /tmp/vsomeip-22
1970-00-01 04:24:47.879286 [info] vSomeIP 3.1.16.1 | (default)
1970-00-01 04:24:56.969346 [info] Stopping vsomeip application "gnss-client" (0022).
1970-00-01 04:24:56.969356 [info] Exiting vsomeip application...
1970-00-01 04:24:56.969366 [info] RELEASE(0022): [00ff.0001]
`

Thoughts?

maingig, 2020-11-05:
I get the same message running the hello_world example, with or without service discovery

Server:
`2020-10-05 14:27:29.623118 [info] Parsed vsomeip configuration in 1ms
2020-10-05 14:27:29.625600 [info] Using configuration file: "/home/fmalouf/Dev/gitlab/build-smart-antenna-services/cfg/gnss-local.json".
2020-10-05 14:27:29.626536 [info] Configuration module loaded.
2020-10-05 14:27:29.627156 [info] Initializing vsomeip application "gnss-service".
2020-10-05 14:27:29.628174 [info] Instantiating routing manager [Host].
2020-10-05 14:27:29.630497 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-10-05 14:27:29.632251 [info] Service Discovery enabled. Trying to load module.
2020-10-05 14:27:29.639408 [info] Service Discovery module loaded.
2020-10-05 14:27:29.640294 [info] Application(gnss-service, 1277) is initialized (11, 100).
2020-10-05 14:27:29.640694 [info] Starting vsomeip application "gnss-service" (1277) using 2 threads I/O nice 255
2020-10-05 14:27:29.643689 [info] main dispatch thread id from application: 1277 (gnss-service) is: 7fecfc42b700 TID: 15502
2020-10-05 14:27:29.643839 [info] shutdown thread id from application: 1277 (gnss-service) is: 7fecfbc2a700 TID: 15503
2020-10-05 14:27:29.644034 [info] Watchdog is disabled!
2020-10-05 14:27:29.647584 [info] OFFER(1277): [1111.2222:0.0] (true)
2020-10-05 14:27:29.648244 [info] io thread id from application: 1277 (gnss-service) is: 7fecfed60740 TID: 15500
2020-10-05 14:27:29.648253 [info] io thread id from application: 1277 (gnss-service) is: 7fecfac28700 TID: 15505
2020-10-05 14:27:29.650747 [info] Listening at /tmp/vsomeip-1277
2020-10-05 14:27:29.651339 [info] vSomeIP 3.1.16.1 | (default)
2020-10-05 14:27:29.652675 [info] Network interface "eno1" state changed: up
2020-10-05 14:27:29.654336 [info] Route "default route (0.0.0.0/0) if: eno1 gw: 172.20.1.1" state changed: up
2020-10-05 14:27:29.655213 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
2020-10-05 14:27:29.655811 [debug] Joining to multicast group 224.224.224.245 from 172.20.1.161
2020-10-05 14:27:29.656371 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 212992
2020-10-05 14:27:29.657336 [info] **Port configuration missing for [1111.2222]. Service is internal.**
2020-10-05 14:27:29.657634 [info] SOME/IP routing ready.
`

lutzbichler, 2020-11-07:
Your configuration file needs to have an entry for the service instance you want to use from another device, like
"services" : [ { "service" : "0x00ff", "instance" : "0x0001", "unreliable" : "30501" } ] if you use the service instance [ff.0001].

maingig, 2020-11-08:
If you are referring to the service config file, I do have this entry.  I stepped through the code it's failing when checking the instance, so it never returns the valid port.  I will post the code tomorrow.

Your wording suggests that you are referring to client.  It seems other posts have suggested you don't need the services entry in the client config file.  Even so, how would that influence the "service is internal" message in the server?

I moved to using the config files shipped with vsomeip and while the local ones work, I still can't get any communication with either the tcp or udp ones.

lutzbichler, 2020-11-09:
If you see a "service is internal" message, the configuration on service side does not match the "offer_service" call. On client side, you do not need the configuration if you use service discovery. If you use service discovery, you need to ensure the route for the multicast address is correctly set to the device/network you are using.

maingig, 2020-11-09:
Your first statement is critical and while the output message is technically correct, this statement made it more obvious as to why the service was internal.  Consider adding something similar to your message output.

The server output now looks correct.  But my client is not connecting - it gets this far:
`1970-00-01 05:26:01.618399 [info] Parsed vsomeip configuration in 4ms
1970-00-01 05:26:01.618409 [info] Using configuration file: "/fDev/gitlab/build-smart-antenna-qnx/install/cfg/helloworld-client.json".
1970-00-01 05:26:01.618409 [info] Initializing vsomeip application "hello_world_client".
1970-00-01 05:26:01.618409 [info] Instantiating routing manager [Host].
1970-00-01 05:26:01.618409 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
1970-00-01 05:26:01.618419 [info] Application(hello_world_client, 5555) is initialized (11, 100).
1970-00-01 05:26:01.618419 [info] Starting vsomeip application "hello_world_client" (5555) using 2 threads I/O nice 255
1970-00-01 05:26:01.618419 [info] Service Discovery disabled. Using static routing information.
1970-00-01 05:26:01.618419 [info] shutdown thread id from application: 5555 (hello_world_client) is: 3
1970-00-01 05:26:01.618419 [info] main dispatch thread id from application: 5555 (hello_world_client) is: 2
1970-00-01 05:26:01.618419 [info] SOME/IP routing ready.
1970-00-01 05:26:01.618419 [info] Watchdog is disabled!
1970-00-01 05:26:01.618419 [info] io thread id from application: 5555 (hello_world_client) is: 5
1970-00-01 05:26:01.618419 [info] io thread id from application: 5555 (hello_world_client) is: 1
1970-00-01 05:26:01.618419 [info] REQUEST(5555): [1111.2222:255.4294967295]
1970-00-01 05:26:01.618429 [info] vSomeIP 3.1.16.1 | (default)
1970-00-01 05:26:01.618429 [info] Listening at /tmp/vsomeip-5555
`

What is the difference between a services entry and a clients entry in the config file?  This is what my config file looks like:

`{
    "unicast":"172.20.1.6",
    "netmask" : "255.255.0.0",
    "logging":
    {
        "level":"debug",
        "console":"true"
    },

    "applications":
    [
        {
            "name":"hello_world_client",
            "id":"0x5555"
        }
    ],

    "services":
    [
        {
            "service":"0x1111",
            "instance":"0x2222",
            "unreliable":"30509"
        }
    ],

    "routing":"hello_world_client",
    "service-discovery":
    {
        "enable":"false"
    }
}
`

maingig, 2020-11-10:
The good news is I got this working with service discovery.  I'm not sure how to get it working otherwise.
I am also only able to get an unreliable service working through service discovery.

lutzbichler, 2020-11-10:
To communicate without SD you need to define the service on client and service side. The client side additionally needs a "unicast" entry with the server sides IP address.

maingig, 2020-11-10:
Ah - another missing piece.  I now have it working without SD.  This does not show up as an option in the documentation except for service groups.  Almost there - I can't get tcp working. I modified the port entry to:

        "reliable": { "port" : "30509", "enable-magic-cookies" : "false" }
Client output:

1970-00-01 03:15:57.572900 [info] Listening at /tmp/vsomeip-5555
Sending: World
1970-00-01 03:15:57.572900 [error] Routing info for remote service could not be found! (5555): [1111.2222.3333] 0001

maingig, 2020-11-16:
Any thoughts about my last comment regarding TCP?

maingig, 2020-11-17:
Okay - last question.  What would cause the client on QNX to think there are multiple routing managers.  I am not running another application I had to set ENABLE_MULTIPLE_ROUTING_MANAGERS=1.

maingig, 2020-11-20:
Last question is not critical.

closed 2020-11-20 by maingig