COVESA/vsomeip #170: service  is not available now, but we may receivce the service data

xiaostang, 2020-11-06:
### the question is the service  is not available now, but we may receivce the service data
* if first handler_type_e::AVAILABILITY data that is not availbale type now is solving.
then, the second handler_type_e::AVAILABILITY data that is available type  is comming, the data should be pending.
function will solve the third data, which may be handler_type_e::MESSAGE. and the data will solved immediately.
the question is the service  is not available now, but we may receivce the service data

[image]

lutzbichler, 2020-12-01:
I am afraid I do not really understand. If I read the code correctly, all handlers are rescheduled until all availability handlers for the same service instance have been executed.

ByteXiaoTang, 2020-12-08:
all availability handlers are sequenced， but messages are not in the same sequence with the availability handlers sequence. we may solve message when service is unavailable solving.

goncaloalmeida, 2023-05-26:
Please provide an example using the last version (3.0.0) of why you need this.

closed 2023-05-26 by goncaloalmeida