COVESA/vsomeip #178: Heap-buffer-overflow in vsomeip_v3::deserializer::look_ahead

fouzhe, 2020-11-18:
There is a heap-buffer-overflow bug and the detailed ASAN report is shown below:
```
=================================================================
==13535==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x606000006538 at pc 0x7f4685133846 bp 0x7f467c2eed80 sp 0x7f467c2eed78
READ of size 1 at 0x606000006538 thread T6 (1277_io01)
    #0 0x7f4685133845 in vsomeip_v3::deserializer::look_ahead(unsigned long, unsigned char&) const /root/vsomeip/implementation/message/src/deserializer.cpp:141:14
    #1 0x7f467dd45057 in vsomeip_v3::sd::message_impl::deserialize_option(vsomeip_v3::deserializer*) /root/vsomeip/implementation/service_discovery/src/message_impl.cpp:373:16
    #2 0x7f467dd44292 in vsomeip_v3::sd::message_impl::deserialize(vsomeip_v3::deserializer*) /root/vsomeip/implementation/service_discovery/src/message_impl.cpp:315:52
    #3 0x7f467dd1e241 in vsomeip_v3::sd::deserializer::deserialize_sd_message() /root/vsomeip/implementation/service_discovery/src/deserializer.cpp:31:44
    #4 0x7f467dd99c64 in vsomeip_v3::sd::service_discovery_impl::on_message(unsigned char const*, unsigned int, boost::asio::ip::address const&, boost::asio::ip::address const&) /root/vsomeip/implementation/service_discovery/src/service_discovery_impl.cpp:1007:28
    #5 0x7f46852a67f6 in vsomeip_v3::routing_manager_impl::on_message(unsigned char const*, unsigned int, vsomeip_v3::endpoint*, boost::asio::ip::address const&, unsigned short, std::pair<unsigned int, unsigned int>, boost::asio::ip::address const&, unsigned short) /root/vsomeip/implementation/routing/src/routing_manager_impl.cpp:1319:37
    #6 0x7f46850fdb44 in vsomeip_v3::udp_server_endpoint_impl::on_message_received(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&, boost::asio::ip::basic_endpoint<boost::asio::ip::udp> const&, std::vector<unsigned char, std::allocator<unsigned char> > const&) /root/vsomeip/implementation/endpoints/src/udp_server_endpoint_impl.cpp:651:39
    #7 0x7f46850eab6b in vsomeip_v3::udp_server_endpoint_impl::on_unicast_received(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&) /root/vsomeip/implementation/endpoints/src/udp_server_endpoint_impl.cpp:496:13
    #8 0x7f46851125b3 in void std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>::operator()<boost::system::error_code const&, unsigned long const&, boost::asio::ip::address const&, void>(boost::system::error_code const&, unsigned long const&, boost::asio::ip::address const&) /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/functional:1131:17
    #9 0x7f46851125b3 in boost::asio::detail::binder3<std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>, boost::system::error_code, unsigned long, boost::asio::ip::address>::operator()() /usr/include/boost/asio/detail/bind_handler.hpp:213:5
    #10 0x7f46851125b3 in void boost::asio::asio_handler_invoke<boost::asio::detail::binder3<std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>, boost::system::error_code, unsigned long, boost::asio::ip::address> >(boost::asio::detail::binder3<std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>, boost::system::error_code, unsigned long, boost::asio::ip::address>&, ...) /usr/include/boost/asio/handler_invoke_hook.hpp:69:3
    #11 0x7f46851125b3 in void boost_asio_handler_invoke_helpers::invoke<boost::asio::detail::binder3<std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>, boost::system::error_code, unsigned long, boost::asio::ip::address>, std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)> >(boost::asio::detail::binder3<std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>, boost::system::error_code, unsigned long, boost::asio::ip::address>&, std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)>&) /usr/include/boost/asio/detail/handler_invoke_helpers.hpp:37:3
    #12 0x7f4685111a58 in boost::asio::detail::reactive_socket_recvfrom_op_ext<boost::asio::mutable_buffers_1, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, std::_Bind<std::_Mem_fn<void (vsomeip_v3::udp_server_endpoint_impl::*)(boost::system::error_code const&, unsigned long, boost::asio::ip::address const&)> (std::shared_ptr<vsomeip_v3::udp_server_endpoint_impl>, std::_Placeholder<1>, std::_Placeholder<2>, std::_Placeholder<3>)> >::do_complete(boost::asio::detail::task_io_service*, boost::asio::detail::task_io_service_operation*, boost::system::error_code const&, unsigned long) /root/vsomeip/implementation/helper/1.55/boost/asio/detail/reactive_socket_recvfrom_op_ext.hpp:121:7
    #13 0x7f4684eb9890 in boost::asio::detail::task_io_service_operation::complete(boost::asio::detail::task_io_service&, boost::system::error_code const&, unsigned long) /usr/include/boost/asio/detail/task_io_service_operation.hpp:38:5
    #14 0x7f4684eb9890 in boost::asio::detail::epoll_reactor::descriptor_state::do_complete(boost::asio::detail::task_io_service*, boost::asio::detail::task_io_service_operation*, boost::system::error_code const&, unsigned long) /usr/include/boost/asio/detail/impl/epoll_reactor.ipp:651:11
    #15 0x7f468556cdc6 in boost::asio::detail::task_io_service_operation::complete(boost::asio::detail::task_io_service&, boost::system::error_code const&, unsigned long) /usr/include/boost/asio/detail/task_io_service_operation.hpp:38:5
    #16 0x7f468556cdc6 in boost::asio::detail::task_io_service::do_run_one(boost::asio::detail::scoped_lock<boost::asio::detail::posix_mutex>&, boost::asio::detail::task_io_service_thread_info&, boost::system::error_code const&) /usr/include/boost/asio/detail/impl/task_io_service.ipp:372:12
    #17 0x7f468556b503 in boost::asio::detail::task_io_service::run(boost::system::error_code&) /usr/include/boost/asio/detail/impl/task_io_service.ipp:149:10
    #18 0x7f468555104d in boost::asio::io_service::run() /usr/include/boost/asio/impl/io_service.ipp:59:25
    #19 0x7f468554606d in vsomeip_v3::application_impl::start()::$_1::operator()() const /root/vsomeip/implementation/runtime/src/application_impl.cpp:420:27
    #20 0x7f468554606d in void std::_Bind_simple<vsomeip_v3::application_impl::start()::$_1 ()>::_M_invoke<>(std::_Index_tuple<>) /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/functional:1530:18
    #21 0x7f468554606d in std::_Bind_simple<vsomeip_v3::application_impl::start()::$_1 ()>::operator()() /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/functional:1520:16
    #22 0x7f468554606d in std::thread::_Impl<std::_Bind_simple<vsomeip_v3::application_impl::start()::$_1 ()> >::_M_run() /usr/lib/gcc/x86_64-linux-gnu/5.4.0/../../../../include/c++/5.4.0/thread:115:13
    #23 0x7f4683994c7f  (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xb8c7f)
[... cut ...]
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==13535==ABORTING
```

attritionorg, 2020-11-18:
@fouzhe Can you describe how this is triggered?

cfanatic, 2020-11-25:
@fouzhe: I would be interested as well. If possible, could you provide the source code?

fouzhe, 2020-12-06:
@attritionorg @cfanatic  Sorry for late!
`notify.json`:
```
{
    "unicast" : "172.19.0.2",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "service-sample",
            "id" : "0x1277"
        },
        {
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509",
            "multicast" :
            {
                "address" : "224.225.226.233",
                "port" : "32344"
            },
            "events" :
            [
                {
                    "event" : "0x0777",
                    "is_field" : "true",
                    "update-cycle" : 2000
                },
                {
                    "event" : "0x0778",
                    "is_field" : "true",
                    "update-cycle" : 0
                },
                {
                    "event" : "0x0779",
                    "is_field" : "true"
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0x4455",
                    "events" : [ "0x777", "0x778" ]
                },
                {
                    "eventgroup" : "0x4465",
                    "events" : [ "0x778", "0x779" ],
                    "is_multicast" : "true"
                },
                {
                    "eventgroup" : "0x4555",
                    "events" : [ "0x777", "0x779" ]
                }
            ]
        }
    ],
    "routing" : "service-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```
1. Parameters
```
VSOMEIP_CONFIGURATION=./notify.json VSOMEIP_APPLICATION_NAME=service-sample ./notify-sample
```
2. It is [notify-sample.cpp](https://github.com/GENIVI/vsomeip/blob/master/examples/notify-sample.cpp)
3. Commit version: `46e6856a1a61f4b24e29c1f2d3ed0b4ee7da0ede`

kheaactua, 2022-02-11:
Any followup on this?  Was it an actual bug?  I've had some crashes that sometimes suggested `look_ahead` was a culprit.

closed 2023-05-26 by goncaloalmeida