COVESA/vsomeip #180: [warning] sd::get_eventgroup_reliability: couldn't determine eventgroup reliability type for [07e3.1f53.a564] using reliability type:  0003

maingig, 2020-11-19:
This warning is output by the client.
I have tried different entries in the configuration file(s) with no luck.  How do you mitigate this message?

maingig, 2020-11-19:
The only way to remove this message is to have the same serivicegroups entry in the client json file which appears in the server json file.  The example config files don't show this but they do have a clients entry.  How does the clients entry differ from the services entry? This entry is from the vsomeip-tcp-client.json file:

`    "clients" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "reliable" : [ "41234" ]
        }
    ],
`

What does the reliable entry mean?  The 41234 is nowhere to be found in the vsomeip-tcp-service.json file.

lutzbichler, 2020-11-30:
Which versions are you using? Could you please provide full logs of client & server as well as the configuration?

maingig, 2020-12-01:
vsomeip version: 3.1.16.1, 
commonApi version: 3.1.12.6, 
commonAPI someip version: 3.1.12.17

Note - the client is running on QNX and I have to enable multiple routing managers in order for it to work.

[client-log.txt](https://github.com/GENIVI/vsomeip/files/5624834/client-log.txt)
[service-log.txt](https://github.com/GENIVI/vsomeip/files/5624836/service-log.txt)
[client-config.txt](https://github.com/GENIVI/vsomeip/files/5624839/client-config.txt)
[service-config.txt](https://github.com/GENIVI/vsomeip/files/5624840/service-config.txt)

lutzbichler, 2020-12-03:
This log message is caused by the compatibility layer that is needed to use CommonAPI 3.1 together with vSomeIP-Lib 3.x. I'd expect it to go away after updating to CommonAPI 3.2 & dropping the compatibility layer.

closed 2020-12-04 by lutzbichler