COVESA/vsomeip #183: solve sub, we should send ack firstly and send field data next

ByteXiaoTang, 2020-11-24:
here send data first but in fact, some user will not use. only when recv the sub ack first

[image]

lutzbichler, 2020-11-27:
Please provide a more detailed description. From what is here, we have no idea what could be the problem.

ByteXiaoTang, 2020-11-28:
routing_manager_base::subscribe --> 
[image]

[image]
we notify data to client, after all these have down, we send ack .
 in fact, we should use send_subscribe_ack first. and then send the current data to the client.

lutzbichler, 2020-12-04:
Fixed with v3.1.20.2

closed 2020-12-04 by lutzbichler