COVESA/vsomeip #185: cyclic offer stopped after stop offer while running notify-sample example

Basav2, 2020-11-27:
### Setup :
vsomeip - server
running notify-sample example as per the readme , changed the vsomeip-local.json (unicast ip to 192.168.56.102)
ip-192.168.56.102
linux -16.04

### Fuzz test scenario : 
while configuring the port for service in vsomeip-local.json , the port is configured to be 30490(same as sd port ) trying to test the robustness , where control path and  data path port is configured to be the same.
config in in vsomeip-local.json file   
 "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30490",

As per the functionality of notify-sample example the offer will be sent cyclically followed by stop offer and this cycle continues, but in this case , the sending of offers is stopped after one cycle , where the socket of sending offers is closed after one cycle .

### Logs: 
_2020-10-27 10:49:49.021341 [info] Parsed vsomeip configuration in 1ms
2020-10-27 10:49:49.021529 [info] Using configuration file: "../../config/vsomeip-local.json".
2020-10-27 10:49:49.021752 [info] Configuration module loaded.
2020-10-27 10:49:49.021843 [info] Initializing vsomeip application "service-sample".
2020-10-27 10:49:49.022569 [info] Instantiating routing manager [Host].
2020-10-27 10:49:49.022936 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-10-27 10:49:49.023624 [info] Service Discovery enabled. Trying to load module.
2020-10-27 10:49:49.026066 [info] Service Discovery module loaded.
2020-10-27 10:49:49.026538 [info] Application(service-sample, 1277) is initialized (11, 100).
2020-10-27 10:49:49.027348 [info] REGISTER EVENT(1277): [1234.5678.8778:is_provider=true]
2020-10-27 10:49:49.028498 [info] Starting vsomeip application "service-sample" (1277) using 2 threads I/O nice 255
2020-10-27 10:49:49.029294 [info] OFFER(1277): [1234.5678:0.0] (true)
2020-10-27 10:49:49.031882 [info] Listening at /tmp/vsomeip-1277
Setting event (Length=1).
2020-10-27 10:49:49.033229 [info] Watchdog is disabled!
2020-10-27 10:49:49.035346 [info] io thread id from application: 1277 (service-sample) is: 7fb6506f7740 TID: 11509
2020-10-27 10:49:49.035944 [info] io thread id from application: 1277 (service-sample) is: 7fb63f7fe700 TID: 11515
2020-10-27 10:49:49.036362 [info] vSomeIP 3.1.16.1 | (default)
2020-10-27 10:49:49.036909 [info] Network interface "enp0s8" state changed: up
2020-10-27 10:49:49.037624 [info] main dispatch thread id from application: 1277 (service-sample) is: 7fb64d5ba700 TID: 11512
Application service-sample is registered.
2020-10-27 10:49:49.037689 [info] Route "224.224.224.245/32 if: enp0s8 gw: n/a" state changed: up
2020-10-27 10:49:49.038595 [info] shutdown thread id from application: 1277 (service-sample) is: 7fb64cdb9700 TID: 11513
**2020-10-27 10:49:49.039031 [info] issue debug log : udp_server_endpoint_impl : binding of local address and port -_local- 192.168.56.102:30490**
2020-10-27 10:49:49.040318 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
**2020-10-27 10:49:49.040658 [info] issue debug log : create_server_endpoint : its_endpoint 0x7fb638000b00  port  30490**
2020-10-27 10:49:49.040827 [debug] Joining to multicast group 224.224.224.245 from 192.168.56.102
2020-10-27 10:49:49.041016 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 212992
2020-10-27 10:49:49.048266 [info] SOME/IP routing ready.
Setting event (Length=2).
Setting event (Length=3).
Setting event (Length=4).
Setting event (Length=5).
Setting event (Length=6).
Setting event (Length=7).
Setting event (Length=8).
Setting event (Length=9).
Setting event (Length=1).
2020-10-27 10:49:59.049533 [info] vSomeIP 3.1.16.1 | (default)
2020-10-27 10:49:59.346717 [info] STOP OFFER(1277): [1234.5678:0.0] (true)
Setting event (Length=2).
**2020-10-27 10:49:59.352282 [info] issue debug log : remove_server_endpoint : its_endpoint 0x7fb638000b00  port  30490
2020-10-27 10:49:59.355535 [info] issue debug log : udp_server_endpoint_impl stop() : unicast socket close - 192.168.56.102:30490**
2020-10-27 10:50:09.055546 [info] vSomeIP 3.1.16.1 | (default)
2020-10-27 10:50:09.923156 [info] OFFER(1277): [1234.5678:0.0] (true)
2020-10-27 10:50:09.924467 [info] issue debug log : udp_server_endpoint_impl : binding of local address and port -_local- 192.168.56.102:30490
2020-10-27 10:50:09.925224 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
2020-10-27 10:50:09.925821 [info] issue debug log : create_server_endpoint : its_endpoint 0x7fb640001670  port  30490_

### When looking at logs (custom logs added for better understanding):
After a socket is created for sending of offers with port 30490 , there is no creation of new socket for the service (since both the ports are same) and socket already exists, after stop offer when the socket needs to be removed the offer sending socket will be closed, and a new socket will be created for the service , this causes the offer sending socket to close.
[... 35 lines cut ...]
Setting event (Length=2).
Setting event (Length=3).
Setting event (Length=4).
Setting event (Length=5).
Setting event (Length=6).
Setting event (Length=7).
Setting event (Length=8).
Setting event (Length=9).
Setting event (Length=1).
2020-10-27 10:45:16.015763 [info] vSomeIP 3.1.16.1 | (default)
2020-10-27 10:45:16.391641 [info] STOP OFFER(1277): [1234.5678:0.0] (true)
**2020-10-27 10:45:16.392585 [info] issue debug log : remove_server_endpoint : its_endpoint 0x7f0f40000e70  port  30508
2020-10-27 10:45:16.393457 [info] issue debug log : udp_server_endpoint_impl stop() : unicast socket close - 192.168.56.102:30508**
2020-10-27 10:45:26.201810 [info] vSomeIP 3.1.16.1 | (default)
2020-10-27 10:45:26.909149 [info] OFFER(1277): [1234.5678:0.0] (true)
2020-10-27 10:45:26.911458 [info] issue debug log : udp_server_endpoint_impl : binding of local address and port -_local- 192.168.56.102:30508
2020-10-27 10:45:26.911750 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992
2020-10-27 10:45:26.911939 [info] issue debug log : create_server_endpoint : its_endpoint 0x7f0f40000d30  port  30508_

### When looking at logs in ideal scenario (port configured to be 30508):
Socket for sending offer is opened with 30490 , and for service with 30508 , after stop offer the  service socket with port 30508 is closed and new socket is opened with port 30508 the functionality continues

### Conclusion:
Wireshark traces are also attached for these scenarios, This is a critical issue where the offers being sent are stopped causing other nodes in the network not to receive offers, causing DOS for other nodes in network , so the configuration parser should check for any misconfigurations in port in the json files and alert the user, before going into socket creation to avoid conflicts.
[port_issue.zip](https://github.com/GENIVI/vsomeip/files/5606244/port_issue.zip)

lutzbichler, 2020-11-30:
30490 is an invalid port for offering services and the host being misconfigured stops working as soon as the service is stopped. We could add a guard to the configuration reader to avoid this behavior. However, as the does not happen using a correct configuration I do not see an issue here.

Basav2, 2020-12-01:
Yes agreed, but there has to be a guard in configuration parser  which validates the info provided by user, before actually going into socket creation.

joao-d, 2023-03-07:
@Basav2 Does the problem still occur?

closed 2023-03-14 by goncaloalmeida