COVESA/vsomeip #186: pending_sd_offers_.erase(it) break may error

ByteXiaoTang, 2020-11-28:
std::vector<std::pair<service_t, instance_t>> pending_sd_offers_;
if find one, break used. in fact,  we may erase all in the vector.
[image]

lutzbichler, 2020-12-01:
Pending offers are inserted on "offer_service", if there hasn´t been an offer command already (checked by insert_offer_command).  Thus, pending_sd_offers_ will not contain the same (service, instance) tuple twice.
Am I missing something?

ByteXiaoTang, 2020-12-08:
in fact, we may not check before insert
[image]

goncaloalmeida, 2023-05-26:
Please provide an example using the last version (3.0.0) of why you need this.

closed 2023-05-26 by goncaloalmeida