COVESA/vsomeip #188: Length of a SOME/IP message with UDP transport

mirzachi, 2020-12-03:
Statement from the article[ vsomeip in 10-minutes ](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes):

> As mentioned above the underlying transport protocol can be UDP or TCP. In the UDP case the SOME/IP messages are not fragmented; it can be that more than one message is in one UDP packet but one message cannot be longer than a UDP package can be (up to 1400 Bytes).

Why only up to 1400B? [UDP ](https://en.wikipedia.org/wiki/User_Datagram_Protocol#:~:text=The%20field%20size%20sets%20a,%E2%88%92%2020%20byte%20IP%20header)allows payload up to 65507 B. 

If the UDP datagram is larger than the typical Ethernet MTU (~1500B), then IP can do the (de)fragmentation. 
The only reason why this would not be the case is, if the used TCP/IP network stack (IP layer) is stripped down and does not offer (de)fragmentation functionality.  But in general this is not the case.

lutzbichler, 2020-12-04:
It is 1400, because the maximum length of SOME/IP messages that use UDP as transport protocol is limited to 1416 byte (1400 bytes of payload) within the SOME/IP specification to avoid fragmentation.

closed 2020-12-04 by lutzbichler

mirzachi, 2020-12-04:
Ah, OK. It is because the specification requires using SOME/IP-TP in the case when the SOME/IP message (UDP binding) is larger than 1416B.