COVESA/vsomeip #189: Getting errors building service-example.cpp

Canorvantis, 2020-12-03:
I am currently trying to get into SOME/IP and therefore trying to run the service-example.cpp from https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes . 

My system is a clean Ubuntu 14.04 running in a virtual machine.

I successfully cloned the repo and ran 
$ cd vsomeip
<.>/vsomeip$ mkdir build
<.>/vsomeip$ cd build
<.>/vsomeip/build$ cmake ..
<.>/vsomeip/build$ make

Then I created a separate folder for the example and copied the service-example.cpp into it, as well as created a build-directory as written in the description. 

Now I am trying to write the CMakeLists.txt file.

In the description it says the CMakeLists.txt could look like:

cmake_minimum_required (VERSION 2.8)

set (CMAKE_CXX_FLAGS "-g -std=c++0x")

find_package (vsomeip 2.6.0 REQUIRED)
find_package( Boost 1.55 COMPONENTS system thread log REQUIRED )

include_directories (
    ${Boost_INCLUDE_DIR}
    ${VSOMEIP_INCLUDE_DIRS}
)

add_executable(service-example ../src/service-example.cpp)
target_link_libraries(service-example vsomeip ${Boost_LIBRARIES})

But running this gives me an error that CMake is unable to find vsomeip.

I then changed     

${VSOMEIP_INCLUDE_DIRS} to contain the actual path to the vsomeip folder.

This allowed to run CMake, but not make.

The "make" step still throws errors since it still seems unable to find the necessary libraries.

It still seems like make is not able to link the libraries correctly.

So how do I have to set up the CMakeListst.txt if I have one folder called "vsomeip" in my home directory where I cloned and built SOME/IP and another folder (also in the Home Directory) where I want to put the service-example code and the CMakeLists.txt files and everything that goes with it?

I know this is probably more of a question related to CMake and it's concepts, but I feel the answer could be short and simple for experts like you.

Thank you in advance

alphaduke0, 2020-12-09:
Maybe replacing the find_package command with find_package(vsomeip3) helps.

andrea-ortalda, 2023-01-24:
Hi, I just faced the same issue and adding the "-DCMAKE_PREFIX_PATH=/root/vsomeip" option to your cmake command seems to do the trick.

/root/vsomeip is where I cloned the vsomeip directory.

This actually set the vsomeip path (where vsomeip3Config.cmake lives) in the CMAKE_PREFIX_PATH variable.

Hope that this helps you as well.

Best.

Andrea O.

goncaloalmeida, 2023-02-24:
@Canorvantis  Does the problem still occur?

closed 2023-03-03 by BrunoLSilvaCTW

wangjingxiaomidou, 2023-11-09:
I build and install vsome based on the guildline, but meet the same issue. How to fix it?
  By not providing "Findvsomeip.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vsomeip", but
  CMake did not find one.

  Could not find a package configuration file provided by "vsomeip"
  (requested version 2.6.0) with any of the following names:

    vsomeipConfig.cmake
    vsomeip-config.cmake

  Add the installation prefix of "vsomeip" to CMAKE_PREFIX_PATH or set
  "vsomeip_DIR" to a directory containing one of the above files.  If
  "vsomeip" provides a separate development package or SDK, be sure it has
  been installed.

dappchandu, 2024-02-15:
> I build and install vsome based on the guildline, but meet the same issue. How to fix it? I have already set the PATH, using export CMAKE_MODULE_PATH=/etc/vsomeip/lib ![image](https://private-user-images.githubusercontent.com/58726252/281694653-f0555e70-3cb8-4637-a815-714c4b2fef6b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwMDg2OTcsIm5iZiI6MTcwODAwODM5NywicGF0aCI6Ii81ODcyNjI1Mi8yODE2OTQ2NTMtZjA1NTVlNzAtM2NiOC00NjM3LWE4MTUtNzE0YzRiMmZlZjZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE1VDE0NDYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTg2YzM2ZmZlNDM2NTUyMzA5YTUyYTU1YjBlNzM4OWRkMGFmY2U4NGZjZTcyOGQ4ZjlkZGMzMDM1OGFmOWZjZDcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.uP86FuRGZ755m9XUdKvHyr_RozVmJGgFZ7W7WeUBT2g) ![image](https://private-user-images.githubusercontent.com/58726252/281694948-7e3c9c4d-b569-430c-b9b5-6dcf039744e1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDgwMDg2OTcsIm5iZiI6MTcwODAwODM5NywicGF0aCI6Ii81ODcyNjI1Mi8yODE2OTQ5NDgtN2UzYzljNGQtYjU2OS00MzBjLWI5YjUtNmRjZjAzOTc0NGUxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjE1VDE0NDYzN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQwM2M0MWQzNjdjZjI2NjYxYzljMjlhNDQ1MDc5YjM0NDM5YzU2ZTc5MDEzMjVhYmQ4YTgzNTkyYzM2NTNiNjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.els1NvjCAzDZIh3TUuTec8fFG638KkgLDETVjntHzN4)
> 
> By not providing "Findvsomeip.cmake" in CMAKE_MODULE_PATH this project has asked CMake to find a package configuration file provided by "vsomeip", but CMake did not find one.
> 
> Could not find a package configuration file provided by "vsomeip" (requested version 2.6.0) with any of the following names:
> 
> ```
> vsomeipConfig.cmake
> vsomeip-config.cmake
> ```
> 
> Add the installation prefix of "vsomeip" to CMAKE_PREFIX_PATH or set "vsomeip_DIR" to a directory containing one of the above files. If "vsomeip" provides a separate development package or SDK, be sure it has been installed.

are you resolved this error because i'm also getting same error