COVESA/vsomeip #190: second_address_test_slave_service_tcp.json not exits.

ningxiao77, 2020-12-07:
in test/second_address_tests/conf/second_address_test_slave_starter.sh.in , need second_address_test_slave_service_tcp.json

but not exits file .

lutzbichler, 2020-12-11:
The json is generated from the .json.in during the build process.

closed 2020-12-14 by lutzbichler

ningxiao77, 2020-12-15:
but the test/CMakeLists.txt not exists the operation。

please double check the issue, thks.

lutzbichler, 2020-12-15:
Lines 3014/3015.

ningxiao77, 2020-12-17:
thanks for your resp.
 but,  lines 3014/3015， just for _udp.json ，not for _tcp.json .

pls check again ,thanks.

""""
  detai below from  https://github.com/GENIVI/vsomeip/blob/master/test/CMakeLists.txt 

  set(TEST_SECOND_ADDRESS_MASTER_SERVICE_UDP_CONFIG_FILE ${TEST_SECOND_ADDRESS_NAME}_master_service_udp.json)
  configure_file(
        ${PROJECT_SOURCE_DIR}/test/second_address_tests/conf/${TEST_SECOND_ADDRESS_MASTER_SERVICE_UDP_CONFIG_FILE}.in
        ${PROJECT_SOURCE_DIR}/test/second_address_tests/${TEST_SECOND_ADDRESS_MASTER_SERVICE_UDP_CONFIG_FILE}
        @ONLY)

'"""""