COVESA/vsomeip #191: Crash: client run at windows system ,server run at unbuntu system

caidy2018, 2020-12-10:
I got a Crash at sentence "std::lock_guard<std::mutex> its_lock(multicast_mutex_) at udp_server_endpoint_impl::join",Vistual Studio 2019 's stack stay at _MyMutex.lock() when crash occur. below is test step:

Step1:
Run a application with vsomeip at unbuntu side as server, config file is server.txt
[server.txt](https://github.com/GENIVI/vsomeip/files/5673634/server.txt)

[client.txt](https://github.com/GENIVI/vsomeip/files/5673633/client.txt)

Step2:
Run another application with vsomeip at windows side as client. config file is client.txt

Phenomenon:
after a while, the windows console print "Didn't receive a multicast SD message for", then Vistual Studio 2019 crash at funtion udp_server_endpoint_impl::join()

caidy2018, 2020-12-11:
In Version 3.1.20.1 have been fix "Do not lock the multicast mutex twice" , but I update vsomeip to 3.1.20.2, the issue still exist.

caidy2018, 2020-12-15:
I found can avoid the windows crash when can't receive multicast message.

modify service_discovery_impl::on_last_msg_received_timer_expired by below:
befer:
#ifndef ANDROID
            reinterpret_cast<udp_server_endpoint_impl*>(
                    endpoint_.get())->join(sd_multicast_);
#else
            dynamic_cast<udp_server_endpoint_impl*>(
                    endpoint_.get())->join(sd_multicast_);
#endif

after:
#if defined(_WIN32) || defined(ANDROID)
                    dynamic_cast<udp_server_endpoint_impl*>(
                            its_service_endpoint.get())->join(sd_multicast_);
#else
                    reinterpret_cast<udp_server_endpoint_impl*>(
                            its_service_endpoint.get())->join(sd_multicast_);
#endif

miaozzZ, 2021-01-20:
i got the same issue.

VG-Alexis, 2021-07-07:
Hi, Same here (execpt that server is on window and client on embedded linux),
The proposed solution seems to work however I think that there is also another place where the issue is.

- service_discovery_impl::start()
- service_discovery_impl::on_last_msg_received_timer_expired()

I finally make different modification, I simply removed the #ifdef and only use dynamic_cast for these two functions which is working.

I'm wondering why there is a reinterpret_cast here ? is it for runtime optimisation reason ?

closed 2023-05-26 by goncaloalmeida