COVESA/vsomeip #193: Cannot load symbol : vsomeip_plugin_init : /usr/lib64/libvsomeip3-e2e.so.3: undefined symbol: vsomeip_plugin_init

ningxiao77, 2020-12-15:
run case :: e2e_tests ,will report the error ::vsomeip_plugin_init : /usr/lib64/libvsomeip3-e2e.so.3: undefined symbol: vsomeip_plugin_init

the detail::
# ./e2e_test_external_master_slave_start.sh  172.16.14.249 172.16.14.250
./e2e_test_external_master_slave_start.sh  172.16.14.249 172.16.14.250
e2e_test_client VSOMEIP_CONFIGURATION::e2e_test_client_external.json_1
e2e_test_service SLAVE_VSOMEIP_CONFIGURATION::e2e_test_service_external.json_1
already connected to 172.16.14.250:9555
33 KB/s (1645 bytes in 0.048s)
e2e_test_client PID:10253
*******************************************************************************
*******************************************************************************
** Please now run:
** e2e_test_external_slave_start.sh
** from an external host to successfully complete this test.
**
** You probably will need to adapt the 'unicast' settings in
** e2e_test_service_external.json and
** e2e_test_client_external.json to your personal setup.
*******************************************************************************
*******************************************************************************
will run in slave cmd:: ./e2e_test_external_slave_start.sh e2e_test_service_external.json_1
wait for e2e_test_client::10253
already connected to 172.16.14.250:9555
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from someip_e2e_test
[ RUN      ] someip_e2e_test.basic_subscribe_request_response
2020-11-15 13:54:39.343356 [info] Parsed vsomeip configuration in 1ms
2020-11-15 13:54:39.345223 [info] Using configuration file: "e2e_test_client_external.json_1".
2020-11-15 13:54:39.346110 [info] Initializing vsomeip application "client-sample".
2020-11-15 13:54:39.347522 [info] Instantiating routing manager [Host].
2020-11-15 13:54:39.348441 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
2020-11-15 13:54:39.350122 [info] Service Discovery enabled. Trying to load module.
2020-11-15 13:54:39.407162 [info] Service Discovery module loaded.
2020-11-15 13:54:39.414752 [info] E2E protection enabled.
2020-11-15 13:54:39.422540 [info] Cannot load symbol : vsomeip_plugin_init : /usr/lib64/libvsomeip3-e2e.so.3: undefined symbol: vsomeip_plugin_init
2020-11-15 13:54:39.423675 [info] Application(client-sample, 1255) is initialized (11, 100).
2020-11-15 13:54:39.425289 [info] Starting...
2020-11-15 13:54:39.425791 [info] Starting vsomeip application "client-sample" (1255) using 2 threads I/O nice 255
2020-11-15 13:54:39.430505 [info] main dispatch thread id from application: 1255 (client-sample) is: 7f99d101d0 TID: 10260
2020-11-15 13:54:39.433190 [info] Watchdog is disabled!
2020-11-15 13:54:39.435770 [info] Client 0x1255 : Service [1234.5678] is NOT available.
2020-11-15 13:54:39.440077 [info] REQUEST(1255): [1234.5678:0.4294967295]
2020-11-15 13:54:39.442304 [info] Avoid trigger SD find-service message for local service/instance/major/minor: 1234/5678/0/4294967295
2020-11-15 13:54:39.442408 [info] io thread id from application: 1255 (client-sample) is: 7f9adcf620 TID: 10253
2020-11-15 13:54:39.444243 [info] shutdown thread id from application: 1255 (client-sample) is: 7f9950f1d0 TID: 10261
2020-11-15 13:54:39.442713 [info] io thread id from application: 1255 (client-sample) is: 7f93fff1d0 TID: 10263
2020-11-15 13:54:39.449765 [info] vSomeIP 3.1.14.1 | (default)
2020-11-15 13:54:39.458523 [info] Listening at /tmp/vsomeip-1255
2020-11-15 13:54:39.460389 [info] Network interface "br0" state changed: up
2020-11-15 13:54:39.464626 [info] REGISTER EVENT(1255): [1234.5678.8001:is_provider=false]
2020-11-15 13:54:39.465993 [info] REGISTER EVENT(1255): [1234.5678.8002:is_provider=false]
2020-11-15 13:54:39.472089 [info] Route "224.0.0.0/4 if: br0 gw: n/a" state changed: up
2020-11-15 13:54:39.473757 [info] udp_server_endpoint_impl: SO_RCVBUF is: 229376
2020-11-15 13:54:39.474963 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 229376
2020-11-15 13:54:39.475911 [info] SOME/IP routing ready.
2020-11-15 13:54:39.228239 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 229376
2020-11-15 13:54:39.233856 [info] Client 0x1255 : Service [1234.5678] is available.
2020-11-15 13:54:39.234632 [warning] BLOCKING CALL AVAILABILITY(1255): [1234.5678]
2020-11-15 13:54:39.240978 [info] dispatch thread id from application: 1255 (client-sample) is: 7f937fe1d0 TID: 10264
2020-11-15 13:54:39.001592 [info] Application/Client cafe is registering.
2020-11-15 13:54:39.005406 [info] Client [1255] is connecting to [cafe] at /tmp/vsomeip-cafe
2020-11-15 13:54:39.014093 [info] REGISTERED_ACK(cafe)
2020-11-15 13:54:39.020553 [info] OFFER(cafe): [4545.0001:0.0] (true)
2020-11-15 13:54:39.021346 [info] Port configuration missing for [4545.1]. Service is internal.
2020-11-15 13:54:39.026515 [info] REGISTER EVENT(cafe): [6767.0001.8001:is_provider=0:reliability=2]
2020-11-15 13:54:39.028194 [info] REQUEST(cafe): [6767.0001:255.4294967295]
2020-11-15 13:54:40.413941 [info] Application/Client 0101 is registering.
2020-11-15 13:54:40.417549 [info] Client [1255] is connecting to [101] at /tmp/vsomeip-101
[... 382 lines cut ...]
third_party/vsomeip/test/e2e_tests/e2e_test_client.cpp:184: Failure
Value of: event_payloads_custom_profile_[received_responses_counters_[0x8002] % vsomeip_test::NUMBER_OF_MESSAGES_TO_SEND][i]
  Actual: '\xC9' (201)
Expected: dataptr[i]
Which is: '\0'
2020-11-15 13:54:43.374200 [warning] 1255: Received all messages ~> going down!
2020-11-15 13:54:43.685770 [info] Stopping...
2020-11-15 13:54:43.189076 [info] Stopping vsomeip application "client-sample" (1255).
2020-11-15 13:54:43.216920 [info] Client [1255] is closing connection to [cafe]
2020-11-15 13:54:43.217974 [info] RELEASE(cafe): [6767.0001]
2020-11-15 13:54:43.219549 [info] Client [1255] is closing connection to [101]
2020-11-15 13:54:43.228487 [info] RELEASE(0101): [1622.0001]
[  FAILED  ] someip_e2e_test.basic_subscribe_request_response (4793 ms)
[----------] 1 test from someip_e2e_test (4793 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (4793 ms total)
[  PASSED  ] 0 tests.
[  FAILED  ] 1 test, listed below:
[  FAILED  ] someip_e2e_test.basic_subscribe_request_response

 1 FAILED TEST
ash: can't kill pid 10253: No such process
end all test.
#

ningxiao77, 2020-12-15:
[e2e_tests.zip](https://github.com/GENIVI/vsomeip/files/5693801/e2e_tests.zip)
yes， I have modify the sh and json file ，and do same with you
you can check the zip file

fcmonteiro, 2023-04-21:
Are you still having this issue?
Latest version should be passing all tests.
Thanks

closed 2023-05-17 by goncaloalmeida