COVESA/vsomeip #194: how to use the physical Ethernet Port instead of loopback port?

lcysys, 2020-12-16:
I am new to the linux , now i build the example try to manually publish the service into the network, however when i run the executable , i want to use the physical ethernet port , it automatically use the loopback port , i can't change , i have configured the IP address in the configuration file, is there anything i missed ? 
[image]
[image]
[image]

lutzbichler, 2020-12-16:
It looks as if the configuration file is not (correctly) read, because the configuration file defines a routing manager and the log from the application claims that this is not the case. However, the "unicast" configuration variable is the only thing that is needed to be defined in order to select the ethernet device.

lcysys, 2020-12-17:
Thank you for the explanation, got it , :) ,now this should be closed!

closed 2020-12-17 by lcysys