COVESA/vsomeip #195: Wrong sleep time after the last OFFER in the Repetition Phase

splywaczyk, 2020-12-16:
According to **[PRS_SOMEIPSD_00416]** startup sequence should be the following:

**Initial Wait Phase:**

1. Wait for random_delay in Range(INITIAL_DELAY_MIN, INITIAL_DELAY__MAX)
2. Send message (Find Service and Offer Service entries)

**Repetition Phase**

Repeat below steps REPETITIONS_MAX times:
1. Wait 2 ^ (REPETITIONS_MAX - 1) * REPETITIONS_BASE_DELAY
2. Send message (Find Service and Offer Service entries)

**Main Phase**
1. Wait CYCLIC_OFFER_DELAY
2. Send message (Offer Service entries)

This means that for **REPETITIONS_MAX = 5** and **REPETITIONS_BASE_DELAY = 100ms** the startup sequence should look like this:

Time | Request
------------ | -------------
0ms | OFFER sent in the Initial Wait Phase
100ms | 1st OFFER sent in the Repetition Phase
300ms | 2ns OFFER sent in the Repetition Phase
700ms | 3rd OFFER sent in the Repetition Phase
1500ms | 4th OFFER sent in the Repetition Phase
3100ms | 5th OFFER sent in the Repetition Phase
5100ms | 1st OFFER sent in the Main Phase
7100ms | 2nd OFFER sent in the Main Phase

But it looks like this:
[image]

So there is **3200ms** between the **5th OFFER** sent in the Repetition Phase and the **1st OFFER** sent in the Main Phase.

I've checked the `service_discovery_impl.cpp` and there is:
```c++
if (_repetition <= repetitions_max_) {
    // Sent offers, double time to wait and start timer again.
    new_delay = std::chrono::milliseconds(_last_delay * 2);
    repetition = ++_repetition;
```

Which means that for 5th `_repetition` the `new_delay` will be `_last_delay * 2` instead of `cyclic_offer_delay_`

Change the condition to `if (_repetition <= repetitions_max_ - 1)` solves this issue but then time between the 5th OFFER sent in the Repetition Phase and the 1st OFFER sent in the Main Phase is little bit less than `cyclic_offer_delay_`

lutzbichler, 2021-01-12:
According to the state machine (PRS_SOMEIPSD_434) the sequence for "REPETITION_MAX=5" the repetition phase is left after waiting for "2^5 * REPETITION_BASE_DELAY" ms. Additionally, the state machine shows no wait for "CYCLIC_OFFER_DELAY" before sending the first message during main phase but does it in the action of the start transition already.

splywaczyk, 2021-01-12:
Thank you for explanation. Now it is clear.

closed 2021-01-12 by splywaczyk