COVESA/vsomeip #201: its_subscription

ByteXiaoTang, 2020-12-28:
error use of its_subscription when its_subscription is_selective
[image]

lutzbichler, 2021-01-11:
What do you think is the error? The overwriting at least is intentional.

ByteXiaoTang, 2021-01-12:
> What do you think is the error? The overwriting at least is intentional.

why?

ByteXiaoTang, 2021-01-12:
> What do you think is the error? The overwriting at least is intentional.

in fact , the  overwriting will not work

lutzbichler, 2021-01-12:
We create a temporary object to handle the case of unsubscribing a single client from a selective event while others remain subscribed. Therefore we cannot use the original object as it would be removed during the unsubscription while being needed for holding the remaining clients.

closed 2021-01-13 by ByteXiaoTang