COVESA/vsomeip #216: Vlan support

ChezzPlaya, 2021-01-15:
Does this stack supports ethernet messages that contain and require a specific VLAN tag id?

lutzbichler, 2021-01-25:
What do you mean by "supports"?

ChezzPlaya, 2021-01-25:
I have a real server host which e.g. sends out the OFFER SOME/IP messages with a specific VLAN Tag ID. Thus, the simulated client shall send for e.g. an Eventgroup SUBSCRIBE SOME/IP - SD message with the same VLAN Tag that has been provided by the server. Otherwise the SUBSCRIBE message won't reach the host. 

So far, I've observed, that the client simulation sends out the FIND service messages without any VLAN tag. 

Is there a way on adding a VLAN tag?

Ps: I'm currently new to the whole SOME/IP world...

splywaczyk, 2021-01-25:
You configure a VLAN tag on the interface level so you have to configure your client's interface to be in the same network and using the same VLAN tag.
For example if your server is available at **192.168.50.1** with VLAN tag 50 then you need that your client will be in the same network for example 192.168.50.2 also using VLAN tag 50.
Assuming your client has interface **eth0** then you could configure it like this:
```
ip link add link eth0 name eth0.50 address AA:AA:AA:AA:AA:A type vlan id 50
ip addr add 192.168.50.2/24 brd 192.168.50.255 dev eth0.50
ip link set dev eth0.50 up
```
and add in your **vsomeipd.json** on the client side:
```
{
   "unicast" : "192.168.50.2",
...
```
After that you should see something like (`ifconfig` output):
```
eth0.50   Link encap:Ethernet  HWaddr aa:aa:aa:aa:aa:aa  
          inet addr:192.168.50.2  Bcast:192.168.50.255  Mask:255.255.255.0
          UP BROADCAST NOARP MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```
Then when client will send FIND request, then VLAN tag = 50 will be added to the Ethernet frame and should be received by the server.

Note that VLAN tags are part of Layer 2 and SOME/IP is a part of Layer 5 so `vsomeip` doesn't care about VALN tags:
[image]
[Image source](https://www.eenewsautomotive.com/content/ip-and-ethernet-motor-vehicles/page/0/2)

closed 2021-01-28 by ChezzPlaya