COVESA/vsomeip #218: Port configuration missing for []. Service is internal

bularcasergiu, 2021-01-18:
This issue is reproduced using vSomeIP 3.1.7.1, and affects the service side when on the target is running a routing manager:
```
 [info] OFFER(4444): [1111.2222:0.0] (true)
 [info] Port configuration missing for [1111.2222]. Service is internal.
```
If i configure my service application also as routing manager the service is registered correctly.
[service_app_as_routing_manager.txt](https://github.com/GENIVI/vsomeip/files/5821960/service_app_as_routing_manager.txt)
[service_app_log_with_default_routing.txt](https://github.com/GENIVI/vsomeip/files/5821961/service_app_log_with_default_routing.txt)
[service_running_with_default_routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821962/service_running_with_default_routing_manager_conf.txt)
[routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821963/routing_manager_conf.txt)
[routing_manager_log.txt](https://github.com/GENIVI/vsomeip/files/5821964/routing_manager_log.txt)
[service_app_as_routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821965/service_app_as_routing_manager_conf.txt)

I need to start a routing manager because on my targets are running more than one vsomeip application. 
These service needs to communicate with a client deployed on other target.

_Originally posted by @bularcasergiu in https://github.com/GENIVI/vsomeip/issues/53#issuecomment-761065264_

lutzbichler, 2021-01-19:
This message only happens if your routing manager application (routingmanagerd) uses a configuration that does not define a port for the service you want to offer.

bularcasergiu, 2021-01-19:
> This message only happens if your routing manager application (routingmanagerd) uses a configuration that does not define a port for the service you want to offer.

Like that looks the routingmanagerd configuration:
`{
    "unicast": "<eth..>",
    "logging":
    {
        "level":"debug",
        "console":"true",
        "file": 
        {
            "enable" : "true",
            "path" : "/tmp/vsomeip.log"
        },
        "dlt" : "false"
    },
    "routing":"routingmanagerd",
    "service-discovery":
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }`

How should look to define a port for this service
`        "service":"0x1111",
        "instance":"0x2222",
        "unreliable":"30509"` ?

lutzbichler, 2021-01-19:
"services" : [
  {
    "service":"0x1111",
    "instance":"0x2222",
    "unreliable":"30509"
  }
]

bularcasergiu, 2021-01-19:
> "services" : [
> {
> "service":"0x1111",
> "instance":"0x2222",
> "unreliable":"30509"
> }
> ]

and this configuration needs to be added to the routing manager configuration file?

lutzbichler, 2021-01-19:
Yes. In general it is a good idea to let all vsomeip application use the same configuration file.

bularcasergiu, 2021-01-19:
Get this! Somehow the 2.10 version support configuring applications in different files, and that help the deployment. But this can be handled ;)

Tx for help!

yooouxin, 2021-12-21:
@bularcasergiu Have you found a better way to handle multi-configuration files?
Actually you metioned 2.10 support different configuration files,but my test still failed.Routing manager host still report Port configuration missing...

pauloasherring, 2021-12-29:
@yooouxin Apparently, files on the same local tree have higher precedence over system-wide config files (e.g. /etc/vsomeip).
So, one way of handling different config files would be to put each file together with each executable in separate folders.

journey175, 2022-02-28:
It is not flexible for put the offered service info into route`s vsomeip.json file, to support remote service find. I think it`s better to fix the route`s config file for multiple service , is that any method to solve this problem? 
thanks for your reply.

closed 2023-05-26 by goncaloalmeida