COVESA/vsomeip #219: No configuration for request_response_delay

tjpsantiago16, 2021-01-25:
Hi @lutzbichler,

according to the section **4.1.4.2** (Server Answer Behavior) of the SOME/IP-SD specification there should be a configuration to make service discovery delay it's response to multicast SOME/IP-SD messages. (REQUEST_RESPONSE_DELAY, which will be configured with min and max value).

When I checked your code, the only available configuration is "request_response_delay" (without a way to give a min and a max value) and even this configuration is never used in the entirety of the code.

1. Is this feature (REQUEST_RESPONSE_DELAY configuration) planned for the future?
2. How does Service Discovery check if the SOME/IP-SD message is from multicast?

lutzbichler, 2021-01-25:
1. It is currently not planned.
2. It gets the information from the UDP server endpoint which uses two different sockets to receive unicast and multicast messages (because otherwise it cannot bind the unicast socket to a specific network address). Older (< 3.1.7) versions used a boost asio extension to extract the receiver address of an incoming message.

tjpsantiago16, 2021-01-28:
@lutzbichler .

may I know if this feature (REQUEST_RESPONSE_DELAY) will be implemented? If not, may I also know the reasoning behind it?

if we are to try to implement this requirement, where is the best place to delay the response for the offer and find services that is received via a multicast address?

fcmonteiro, 2023-10-20:
Hi @tjpsantiago16, 
The feature was never implemented because it's something we've never needed internally.
There is still not planned implementation of it, but if the community wants to implement it we will accept it.
Thanks

closed 2023-10-20 by fcmonteiro