COVESA/vsomeip #220: Does vsomeip3 configuration support as both client and service at the same time?

rlokanathan, 2021-01-25:
Hi,

I am having problem in communicating to my vsomeip applications between my host and target.
The current situation is, I have three vsomeip applications running in my environment. 
1. Client app (built using commonapi)
2. Service app (built using commonapi)
3. Dummy server app (built using vsomeip)

Basically, how the communication works is my client will request for the service. Then from there my service app will receive the request and then it will send the request to the dummy server app. The server app will send the response to the service app. From there the service app will send the response back to the client app. 

Now the problem I am facing is, my service app not able to get the event back from the dummy server app. Unless if I run all three aps in same machine then all the apps are working fine. 

Current OS: Ubuntu
Vsomeip version: 3.1.7.1
Commonapi verson: 3.20

Attached are my logs from my three apps.
[inc-client-log.txt](https://github.com/GENIVI/vsomeip/files/5865528/inc-client-log.txt)
[inc-service-log.txt](https://github.com/GENIVI/vsomeip/files/5865530/inc-service-log.txt)
[dummy-server-log.txt](https://github.com/GENIVI/vsomeip/files/5865532/dummy-server-log.txt)

**client.json:**

     {
     "unicast":"192.168.2.1",
     "netmask":"255.255.255.0",
     "logging":
     {
         "level":"debug",
         "console":"true",
         "file":{"enable":"true", 
         "path":"/tmp/vsomeip.log"},
	 "dlt":"false"
     },

    "applications":
    [
        {
            "name":"IncClient"
        }
    ],
	"clients":
    [
        {
	    "service":"0x0001",
            "instance":"0x0002",
            "reliable":{"port":"80","enable-magic-cookies":"false"}
        }
    ],

    "routing":"IncClient",
    "service-discovery":
    {
        "enable":"true",
	"multicast":"224.0.0.1",
	"port":"30490",
	"protocol":"udp",
	"initial_delay_min":"10",
        "initial_delay_max":"100",
        "repetitions_base_delay":"200",
        "repetitions_max":"3",
        "ttl":"3",
        "cyclic_offer_delay":"2000",
        "request_response_delay":"1500"
    }
    }

**service.json:**

     {
     "unicast":"192.168.2.40",
     "netmask":"255.255.255.0",
     "network":"multi2",
     "logging":
     {
        "level":"debug",
        "console":"true",
		"file":{"enable":"true", "path":"/tmp/vsomeip.log"},
		"dlt":"false"
     },

    "applications":
    [
	{
	    "name":"IncService"
	}
    ],

    "services":
    [
        {
            "service":"0x0001",
            "instance":"0x0002",
            "reliable":{"port":"80","enable-magic-cookies":"false"}
        }
    ],

    "routing":"IncService",

    "service-discovery":
    {
        "enable":"true",
	"multicast":"224.0.0.1",
	"port":"30490",
	"protocol":"udp",
	"initial_delay_min":"10",
	"initial_delay_max":"100",
	"repetitions_base_delay":"200",
	"repetitions_max":"3",
	"ttl":"3",
	"cyclic_offer_delay":"2000",
	"request_response_delay":"1500"
    }
    }

**dummyserver.json**

     {
     "unicast":"192.168.2.40",
     "netmask":"255.255.255.0",   
     "network":"multi2",
     "logging":
     {
        "level":"debug",
        "console":"true",
		"file":{"enable":"true", "path":"/tmp/vsomeip.log"},
		"dlt":"false"
     },

    "applications":
    [
        {
            "name":"voltageServer"
        }
    ],

    "services":
    [
        {
            "service":"0x0004",
            "instance":"0x0002",
            "reliable":{"port":"80","enable-magic-cookies":"false"}
        }
    ],
    
    "service-discovery":
	{
        "enable":"true",
	"multicast":"224.0.0.1",
	"port":"30490",
	"protocol":"udp",
	"initial_delay_min":"10",
        "initial_delay_max":"100",
        "repetitions_base_delay":"200",
        "repetitions_max":"3",
        "ttl":"3",
        "cyclic_offer_delay":"2000",
        "request_response_delay":"1500"

    }
    }

I am not really sure what I have done wrong in my json configuration file. Does the vsomeip3 configuration file supports to configure the service app as both client and service? I really need help on this as I have spend almost a week debugging this issue.

linenli, 2021-06-23:
In my experience in using vsomeip, it seems to support to configure service and client in the same configuration file.

ims0, 2021-12-08:
support

goncaloalmeida, 2023-02-17:
@rlokanathan support. Does the problem still occur?

goncaloalmeida, 2023-02-23:
Closing the ticket -> if the problem still occurs please reopen the ticket.

closed 2023-02-23 by goncaloalmeida