COVESA/vsomeip #226: vsomeip 3.1.15 cannot receive response from service in Linux

rlokanathan, 2021-02-05:
Hi,

This is my current setup.

Windows:
HelloWorld client app (192.168.2.1)
-	Vsomeip3.1.15
-	Commonapi 3.2.0
-	Commonapi-someip 3.2.0
-	Boost 1.67

Linux:
HelloWorld service app (192.168.2.40)
-	Vsomeip3.1.15
-	Commonapi 3.2.0
-	Commonapi-someip 3.2.0
-	Boost 1.67

Currently, I am having issue to receive the response from the service app. I am using the sample hello world app where the client sends the string "Bob" and then the service will send back the response as "Hello Bob!". Currently, with the commonapi library, I am not getting the reply back from the service. From the pcap log, I can see that the client is sending the request to the service, and service able to receive the request. Then service is trying to send the response to the client. But then the reply is not receiving by the client. 

Here's my pcap:

[pcap](https://paste.pics/BIBKF)

However, when I try the same app in Linux environment it works. Only in Windows it didnt work. And also I tried the vsomeip3 sample helloworld app in the same setup as above without common api. It works actually. I am not sure what is wrong.

Powana, 2021-11-26:
Same issue here.

goncaloalmeida, 2023-02-17:
@rlokanathan this problem still occur?

goncaloalmeida, 2023-02-23:
Closing the ticket -> if the problem still occurs please reopen the ticket.

closed 2023-02-23 by goncaloalmeida

thanhtrung17072001, 2025-10-01:
Hello @goncaloalmeida 
Currently, I got this error when communication between Windows PC (client application) and Linux board (server application). Do you know how to fix this issue? If yes, please help me. Thank you so much.