COVESA/vsomeip #227: FindOfferEntry is not answered by the server on Windows 10

ChezzPlaya, 2021-02-06:
[image]

Configuration:

```
{
    "unicast" : "fe80::60a1:4ef:4234:888e",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true"
    },
    "applications" : 
    [
        {
            "name" : "service-sample",
            "id" : "0x1277"
        },
        {    
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "services" :
    [
        {
            "service" : "0x600E",
            "instance" : "0x0001",
            "unreliable" : "30509",
            "events" : 
            [
                {
                    "event" : "0x0777",
                    "is_field" : "true",
                    "update-cycle" : 5000
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0xC010",
                    "events" : ["0x0777"]
                }
            ]
        }
    ],
    "routing" : "service-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "ff14::4:0",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

Why does the server didn't answer the clients FindService request? I expect an OfferService Reply from `fe80::60a1:4ef:4234:888e `to `fe80::60a1:4ef:4234:8888`.

RBEI-ArunKumar, 2023-04-12:
@ChezzPlaya Could you help me with settings on how you are able to use this library in Windows 10? I managed to compile the sources using Visual Studio 2019 compiler, but when running the examples, I always get below error on Windows 10 OS:

2023-04-12 14:46:38.894062 [error] vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]
2023-04-12 14:46:38.894590 [info] Application(vsomeip_ctrl, ffff) is initialized (11, 100).
2023-04-12 14:46:38.894950 [info] Starting vsomeip application "vsomeip_ctrl" (ffff) using 2 threads
2023-04-12 14:46:38.895327 [error] vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]

ChezzPlaya, 2023-04-12:
@RBEI-ArunKumar Sorry, I'm not interested in using this lib anymore. Maybe you should ask for the devs (good luck for getting a response!)... But in general, I'd recommend to implement SOME/IP simulation yourself, rather then using this.

closed 2023-04-12 by ChezzPlaya

thanhtrung17072001, 2025-10-01:
Hello @ChezzPlaya,
Currently, I got this error when communication between Windows PC and Linux board. Do you know how to fix this issue? If yes, please help me. Thank you so much.