COVESA/vsomeip #229: Didn't insert subscription as subscription doesn't match reliability type

ChezzPlaya, 2021-02-09:
By using the subscribe sample, I get this warning:

`vsomeip_v3::sd::service_discovery_impl::create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [60cc.0002.8008] 2`

My simulation setup is server (real ecu) <-> client (simulated, vsomeip). I'm using this configuration: 

```
{
    "unicast" : "fd53:7cb8:383:e::3aa",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true"
    },
    "applications" : 
    [
        {
            "name" : "service-sample",
            "id" : "0x1277"
        },
        {    
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "servicegroups" :
    [
        {
            "name" : "default",
            "unicast" : "fd53:7cb8:383:e::108",
            "services" :
            [
                {
                    "service" : "0x60cc",
                    "instance" : "0x0002",
                    "reliable" : "29180",
                    "events" :
                    [
                        {
                            "event" : "0x8008",
                            "is_field" : "true",
                            "update-cycle" : 5000,
                            "is_reliable" : "true"
                        }
                    ],
                    "eventgroups" :
                    [
                        {
                            "eventgroup" : "0x8008",
                            "events" : ["0x8008"]
                        }
                    ]
                }
            ]
        }
    ],
    "routing" : "client-sample",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "ff14::4:0",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

The original code of the subscribe sample is unchanged, except the sampleIds.

Why do I get this warning and how to fix this?

lutzbichler, 2021-02-12:
Could you please show the Offer message send by the server?

ChezzPlaya, 2021-02-15:
> Could you please show the Offer message send by the server?

Here is the Offer Service:

[image]

Have I configured the client incorrectly?

ChezzPlaya, 2021-02-17:
Any ideas? This is also reproducible with the default subscribe/notify example when using Ipv6 Addresses on two different laptops.

frankie-zeng, 2021-09-27:
change the example to this, subscribe event when service&instance  valid.
```
void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
        std::cout << "Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is "
            << (_is_available ? "available." : "NOT available.")
            << std::endl;
        if (_is_available) {
            app_->subscribe(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENTGROUP_ID);
        }
    }
```

Sudozh, 2021-12-21:
I also encountered this problem, have you solved it?

ChezzPlaya, 2021-12-21:
> I also encountered this problem, have you solved it?

No, I didn't... Currently, I'm also interested of using this library.

Sudozh, 2021-12-21:
> > I also encountered this problem, have you solved it?
> 
> No, I didn't... Currently, I'm also interested of using this library.

Thanks a lot for you.

Sudozh, 2021-12-26:
I solved this problem with the help of my colleague. But i don't know if it works for you。
If you used CommonAPI, you can check the *.fdpel file.Look for the "SomeIpReliable" keyword to see if its configuration is the same as the json file configuration.
just as :
```
# json
"is_reliable" : "true"
```

```
# *.fdpel
SomeIpReliable = true
```

mohitrao12, 2022-12-06:
> change the example to this, subscribe event when service&instance valid.
>
> ```
> void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
>         std::cout [...]

I tried this but didn't work for me if you know any other solution then please let me know

mohitrao12, 2022-12-06:
> I solved this problem with the help of my colleague. But i don't know if it works for you。 If you used CommonAPI, you can check the *.fdpel file.Look for the "SomeIpReliable" keyword to see if its [...]
where I can find this file # *.fdpel

anaritarodrigues, 2023-09-06:
@ChezzPlaya, , Which version of CommonAPI were you using? 
Also, please add the fdepl files are you using.

ChezzPlaya, 2023-09-06:
@anaritarodrigues Sorry, I'm not using anymore this or the CommonAPI libarary.

closed 2023-09-06 by anaritarodrigues

yl2886, 2024-04-26:
The same issue with running client in qemu simulated continer, have u solved it?