COVESA/vsomeip #231: Error building vsomeip 3.1.20.3 with latest version of boost 1.75.0

dmrauh, 2021-02-24:
Similar to errors described in https://github.com/GENIVI/vsomeip/issues/18, trying to build `vsomeip 3.1.20.3` with `boost 1.75.0` results in build errors. Most likely there were some interface changes in `boost` between `1.74.0` and `1.75.0`. Maybe it's an easy fix and `CMake` has just to deal a bit differently with the newest boost version.

Here are some of the errors, to give you an impression:

```
[ 55%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/utility/src/utility.cpp.o
In file included from /home/dominik/Code/workspace/build/vsomeip-3.1.20.3/implementation/endpoints/src/local_server_endpoint_impl.cpp:17:
/home/dominik/Code/workspace/build/vsomeip-3.1.20.3/implementation/endpoints/src/../include/local_server_endpoint_impl.hpp:15:10: fatal error: boost/asio/local/stream_protocol_ext.hpp: No such file or directory
   15 | #include <boost/asio/local/stream_protocol_ext.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [CMakeFiles/vsomeip3.dir/build.make:154: CMakeFiles/vsomeip3.dir/implementation/endpoints/src/local_server_endpoint_impl.cpp.o] Error 1
make[4]: *** Waiting for unfinished jobs....
In file included from /home/dominik/Code/workspace/build/vsomeip-3.1.20.3/implementation/endpoints/src/udp_server_endpoint_impl.cpp:18:
/home/dominik/Code/workspace/build/vsomeip-3.1.20.3/implementation/endpoints/src/../include/udp_server_endpoint_impl.hpp:10:10: fatal error: boost/asio/ip/udp_ext.hpp: No such file or directory
   10 | #include <boost/asio/ip/udp_ext.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[4]: *** [CMakeFiles/vsomeip3.dir/build.make:271: CMakeFiles/vsomeip3.dir/implementation/endpoints/src/udp_server_endpoint_impl.cpp.o] Error 1
/home/dominik/Code/workspace/build/vsomeip-3.1.20.3/implementation/endpoints/src/server_endpoint_impl.cpp:14:10: fatal error: boost/asio/ip/udp_ext.hpp: No such file or directory
   14 | #include <boost/asio/ip/udp_ext.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

What would be necessary to make `vsomeip` compatible with `boost 1.75.0`?

lutzbichler, 2021-03-05:
Try changing lines 167ff in CMakeLists.txt to:

if (${VSOMEIP_BOOST_VERSION} GREATER 107500)
message( ERROR "boost version ${VSOMEIP_BOOST_VERSION} is not (yet) supported. Latest supported version is 1.75.0" )
elseif(${VSOMEIP_BOOST_VERSION} GREATER 107400)

dmrauh, 2021-03-12:
Thank you! That solved the problem.

closed 2021-03-12 by dmrauh