COVESA/vsomeip #235: Running hello world example

EliasVerstappe, 2021-03-02:
Hi, 

My problem is related to https://github.com/GENIVI/vsomeip/issues/56
For some reason, that issue is closed without being resolved.

I'm using SOME/IP for a school, but I'm having trouble getting the basic hello world example working.
My setup is as follows. I'm using Ubuntu 18.04 in WSL. After installing all dependencies, I was able to build the vsomeip stack.
This was achieved by following the steps from the README file.

(Notice that I provide path information in my code segments to illustrate in which directories I execute things)

```
../wsl$ git clone https://github.com/GENIVI/vsomeip.git
../wsl$ cd vsomeip

../wsl/vsomeip$ mkdir build
../wsl/vsomeip$ cd build

../wsl/vsomeip/build$ cmake -DENABLE_SIGNAL_HANDLING=1 ..
../wsl/vsomeip/build$ make
../wsl/vsomeip/build$ sudo make install
```

Next, I tried to build the examples. Here is what I did:
```
../wsl/vsomeip$ cd examples/hello_world
../wsl/vsomeip/examples/hello_world$ mkdir build
../wsl/vsomeip-examples/hello_world$ cd build
../wsl/vsomeip/examples/hello_world/build$ cmake ..
../wsl/vsomeip/examples/hello_world/build$ make
```

However, the `cmake ..` command did not execute successfully (so I did not execute the subsequent `make` command). This is the error message I got:
```
CMake Error at CMakeLists.txt:35 (find_package):
  find_package called with incorrect number of arguments

 was not found. Please specify vsomeip_DIR
-- Configuring incomplete, errors occurred!
```

I was able to fix this problem by reinstalling the vsomeip stack to a custom path (namely '../wsl/vsomeip/vsomeip_stack'), and explicitly setting the VSOMEIP_NAME variable.
Before running the `cmake ..` command to build the example, I edited '../wsl/vsomeip/examples/hello_world/CMakeList.txt' file by adding `set(VSOMEIP_NAME "../wsl/vsomeip/vsomeip_stack/lib/cmake/vsomeip3/vsomeip3") under the line where the C++11 flag is set.

#### **`../wsl/vsomeip/examples/hello_world/CMakeList.txt`**
```
# Copyright (C) 2015-2017 Bayerische Motoren Werke Aktiengesellschaft (BMW AG)
# This Source Code Form is subject to the terms of the Mozilla Public
# License, v. 2.0. If a copy of the MPL was not distributed with this
# file, You can obtain one at http://mozilla.org/MPL/2.0/.

cmake_minimum_required (VERSION 2.8.7)
project (vSomeIPHelloWorld)

find_package(Threads REQUIRED)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -std=c++11")
set(VSOMEIP_NAME "/mnt/c/Users/Elias/wsl/vsomeip/vsomeip_stack/lib/cmake/vsomeip3/vsomeip3") <--- Line I added

include_directories(${VSOMEIP_INCLUDE_DIRS})

add_library(vsomeip_hello_world_service INTERFACE)
target_sources(vsomeip_hello_world_service INTERFACE
    "${CMAKE_CURRENT_SOURCE_DIR}/hello_world_service.hpp"
)
... rest of the file ...
```

After adding that line, the `cmake ..` command executed successfully. Building the example could be finished by running `make`.

At this point I would guess the hello world example was compiled correctly.
Before running the examples, I edited the `helloworld-local.json` file, changing the "unicast"-key's value to "127.0.0.1" (localhost).

Now I wanted to run the examples by running the service in one terminal and running the client in another terminal.
This did not work for me. 

I ran the following command:

```
../wsl/vsomeip/examples/hello_world/build$ VSOMEIP_CONFIGURATION=../helloworld-local.json VSOMEIP_APPLICATION_NAME=hello_world_service ./hello_world_service
```

But this returns an error in my terminal:
```
Configuration module could not be loaded!
```

The same is true for the client. 
I can't find a lot of information on the internet about this. In my experience, documentation is sparse and incomplete. I've built the documentation using `make doc` in the '../wsl/vsomeip/build' directory, but could not find any troubleshooting info there either.

Can anyone point me to some troubleshooting guides or help me solve my problem?
Thanks in advance, Elias

cjkoenig, 2021-03-03:
Hi,

I guess this error 

> `Configuration module could not be loaded!`

can be fixed by running

`sudo ldconfig`

to refresh the library cache of your linux system. You can check if the library is included in the cache (and can be found when you run your program) with this command:

`ldconfig -p | grep libvsomeip`

EliasVerstappe, 2021-03-04:
That does indeed seem to fix the issue, thank you very much!

closed 2021-03-04 by EliasVerstappe

sateesh12, 2021-05-21:
> Hi,
> 
> I guess this error
> 
> > `Configuration module could not be loaded!`
> 
> can be fixed by running
> 
> `sudo ldconfig`
> 
> to refresh the library cache of your linux system. You can check if the libraries are included in the cache (and can be found when you run your program) with this command:
> 
> `ldconfig -p | grep libvsomeip`

This works, thank you very much.

ByteXiaoTang, 2021-05-21:
The lib path should be loaded

mohitrao12, 2022-11-24:
> Hi,
>
> I guess this error
>
>
> can be fixed by running
>
> `sudo ldconfig`
>
> to refresh the library cache of your linux system. You can check if the libraries are included in the cache (and can be found when [...]

worked bro !! thank you so much!!

muellerc, 2023-09-09:
> > Hi,
> > I guess this error
> > > `Configuration module could not be loaded!`
> > 
> > 
> > can be fixed by running
> > `sudo ldconfig`
> > to refresh the library cache of your linux system. You can check if the libraries are included in the cache (and can be found when you run your program) with this command:
> > `ldconfig -p | grep libvsomeip`
> 
> I just started working on vsomeip3.3, but I am getting this error and it's not getting resolved after running these commands. Please help.

Same here... :-/

tomcat023, 2023-09-12:
I had the same issue when I was trying to install this on the Raspberry PI 4, running raspian.
I built vsomeip in its directory in the: "build folder"

`cd build`

`make install`

It is not good enough to just make the libraries, you have to install them. If you did not, then it can't load anything! I only say that because I just did it.