COVESA/vsomeip #240: Question about e2e profile 4

xbroquer, 2021-03-23:
Hello,

First of all thank you very much for implementing e2e profile 4 protection!

I have a question about e2e header generation and more precisely on data_id , for SWS_E2E_00366 req :

https://github.com/GENIVI/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/e2e_protection/src/e2e/profile/profile04/protector.cpp#L37

https://github.com/GENIVI/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/e2e_protection/include/e2e/profile/profile04/profile_04.hpp#L15

I were not able to find the Req that impose to add the instance ID in MSB of data id. 

Here the document I used to check the implementation:
https://www.autosar.org/fileadmin/user_upload/standards/classic/4-2/AUTOSAR_SWS_E2ELibrary.pdf

Do you have some detail why you added the instance id ?

Thanks,
Xavier.

xbroquer, 2021-06-21:
Follow up :)

@lutzbichler Can you please clarify this particular point ? 
I just forked this repo to only remove the instance_id in order to get e2e profile working on my side.

lutzbichler, 2021-06-22:
The instance ID was added to avoid using the same data id for different instances of the same service. We will make this optional.

closed 2021-06-22 by lutzbichler

xbroquer, 2021-06-24:
thanks , having this optional will be fine

richardsanqi, 2026-09-23:
Hi @lutzbichler,

  In #240, you mentioned that including the instance ID in the DataID would be made optional. Has this option been implemented? If so, which release introduced it, and what configuration parameter or API enables it?

  We are facing an interoperability issue with an ECU that expects the configured DataID to be transmitted unchanged:

  - SOME/IP instance ID: 0x0001
  - Configured P04 DataID: 0x0000000e
  - Expected wire DataID: 0x0000000e
  - Actual wire DataID: 0x0100000e

  The P04 implementation we inspected constructs the DataID as:

  uint32_t its_data_id(uint32_t(_instance) << 24 | config_.data_id_);

  The receiving ECU reports matching CRC and length values, but a DataID mismatch. Our customer requires the complete 32-bit wire DataID to remain 0x0000000e.

  We could not find an option to disable instance ID encoding in the P04 configuration parser or documentation.

  If this option is not available, what is the recommended approach without modifying the upstream library source? Would a custom E2E provider loaded through VSOMEIP_E2E_PROTECTION_MODULE be the supported solution?

  We would prefer a per-protected-event option that preserves the existing behavior by default and allows the configured 32-bit DataID to be used unchanged.

  Thanks for your guidance.

brenkem, 2026-10-19:
Hi.

Following up on @richardsanqi — same issue here with vsomeip 3.7.2 against a series production ECU.

AUTOSAR_PRS_E2EProtocol [PRS_E2E_00366] (Figure 6.33): "Copy 4-byte Config->DataID to bytes Data[Offset+4...Offset+7] in Big Endian order". There is no instance component. Uniqueness across instances is a matter of system configuration [PRS_E2E_UC_00327]. With **instance << 24 | config_.data_id_**, vsomeip overwrites the configured DataID. Conformant receivers reject the frame, and DataIDs ≥ 0x01000000 are silently corrupted.

The option promised above has not landed. protector.cpp and checker.cpp are unchanged on master.

Related: #805. The 1..255 instance limit is a direct consequence of this composition (one byte), not an independent design constraint.

Note: e2e_profile_04_test_* hardcodes the composed value (0x0100002d for data_id 0x2d), so it has to be updated with any fix.

I am working on a patch. Could this issue be reopened? Or do you prefer a PR?