COVESA/vsomeip #242: CMake Error at examples/hello_world

FebruaryFB, 2021-04-04:
Hi,

I'm having trouble on make the  hello world example: Unknown CMake command "target_sources".
 Could you help me?  thank you very much.

My setup is as follows. 

```
FebruaryFB:~/vsomeip/vsomeip-master$ mkdir build
FebruaryFB:~/vsomeip/vsomeip-master$ cd build
FebruaryFB:~/vsomeip/vsomeip-master/build$ cmake ..

-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.55.0
-- Found the following Boost libraries:
--   system
--   thread
--   filesystem
-- Boost_LIBRARY_DIR not empty using it: /usr/lib/x86_64-linux-gnu
-- Using boost version: 105500
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.26") 
-- checking for module 'automotive-dlt >= 2.11'
--   package 'automotive-dlt >= 2.11' not found
-- checking for module 'libsystemd'
--   package 'libsystemd' not found
-- Systemd was not found, watchdog disabled!
-- Default configuration folder: /etc/vsomeip
-- Default configuration file: /etc/vsomeip.json
Predefined base path: /tmp
Predefined unicast address: 127.0.0.1
Predefined diagnosis address: 0x01
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.6") 
dot found
asciidoc found
source-highlight found
CMake Error at examples/hello_world/CMakeLists.txt:16 (target_sources):
  Unknown CMake command "target_sources".

-- Configuring incomplete, errors occurred!
See also "/home/feibo/vsomeip/vsomeip-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/feibo/vsomeip/vsomeip-master/build/CMakeFiles/CMakeError.log".

FebruaryFB:~/vsomeip/vsomeip-master/build$ cmake -version
cmake version 2.8.12.2
```

MrBeepoint, 2021-04-30:
command target sources was added in CMake version 3.1 (https://cmake.org/cmake/help/latest/command/target_sources.html). You are running version 2.8.12.2, so you need to upgrade CMake if you want to use the existing config.

goncaloalmeida, 2023-02-17:
Closing ticket: Ticket already answered.

closed 2023-02-17 by goncaloalmeida