COVESA/vsomeip #245: cmake with IPsec plugin not supported

elvin03, 2021-05-03:
Hi,

I'm having trouble on compilation of the IPsec plugin.
According to the CHANGES at v2.11.0, IPsec plugin is added and the compilation of the plugin has to be explicitly enabled when calling cmake. Example: cmake -DPLUGIN_IPSEC_BUILD=1 ..

Therefore when i execute cmake -DPLUGIN_IPSEC_BUILD=1 .. , cmake gives me the warning where the variables were not used by the project.

`user@apertispro:~/vsomeip/build$ cmake -DPLUGIN_IPSEC_BUILD=1 ..`
`-- The C compiler identification is GNU 8.3.0 `
`-- The CXX compiler identification is GNU 8.3.0`
`-- Check for working C compiler: /usr/bin/cc`
`-- Check for working C compiler: /usr/bin/cc -- works`
`-- Detecting C compiler ABI info`
`-- Detecting C compiler ABI info - done`
`-- Detecting C compile features`
`-- Detecting C compile features - done`
`-- Check for working CXX compiler: /usr/bin/c++`
`-- Check for working CXX compiler: /usr/bin/c++ -- works`
`-- Detecting CXX compiler ABI info`
`-- Detecting CXX compiler ABI info - done`
`-- Detecting CXX compile features`
`-- Detecting CXX compile features - done`
`-- Setting build type to 'RelWithDebInfo' as none was specified.`
`-- Looking for pthread.h`
`-- Looking for pthread.h - found`
`-- Looking for pthread_create`
`-- Looking for pthread_create - not found`
`-- Looking for pthread_create in pthreads`
`-- Looking for pthread_create in pthreads - not found`
`-- Looking for pthread_create in pthread`
`-- Looking for pthread_create in pthread - found`
`-- Found Threads: TRUE  `
`-- Boost version: 1.67.0`
`-- Found the following Boost libraries:`
`--   system`
`--   thread`
`--   filesystem`
`--   chrono`
`--   date_time`
`--   atomic`
`-- Using boost version: 106700`
`-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") `
`-- Checking for module 'automotive-dlt >= 2.11'`
`--   Found automotive-dlt , version 2.18.5`
`-- Checking for module 'libsystemd'`
`--   Found libsystemd, version 241`
`-- Default configuration folder: /etc/vsomeip`
`-- Default configuration file: /etc/vsomeip.json`
`Predefined base path: /tmp`
`Predefined unicast address: 127.0.0.1`
`Predefined diagnosis address: 0x01`
`-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) `
`CMake Warning at CMakeLists.txt:525 (message):`
`  Doxygen is not installed.  Documentation can not be built.`

`CMake Warning at CMakeLists.txt:564 (message):`
`  asciidoc is not installed.  Readme can not be built.`

`-- GTEST_ROOT is not defined. For building the tests the variable`
`             GTEST_ROOT has to be defined. Tests can not be built.`
`-- Configuring done`
`-- Generating done`
`CMake Warning:`
`Manually-specified variables were not used by the project:`

    PLUGIN_IPSEC_BUILD

`-- Build files have been written to: /home/user/vsomeip/build`

Can you help me on this? thank you very much

elvin03, 2021-06-04:
Does anyone able to compile vsomeip with IPsec plugin?

lutzbichler, 2021-06-22:
IPsec plugin was removed from the package.

closed 2021-06-22 by lutzbichler