COVESA/vsomeip #250: The interval between two messages

yooouxin, 2021-05-16:
I find that there has a interval between two messages,even if there is no delay between two transmissions.
I write a very sample code for test latency,application will send a message continuously,but in fact,it has a interval between two messages about 2ms
[image]
[image]

But when I insert a delay between transmissions,it comes ok.When I send a message in a 100ms cycle, the result of grabbing the packet is also 100ms.

I want to know why this happened.
Thanks

ByteXiaoTang, 2021-05-17:
Wireshark本身带刷新采集周期

yooouxin, 2021-05-17:
> Wireshark本身带刷新采集周期

并不是这个原因，裸boost库或socket是不会到2ms这个级别的，应该是vsomeip底层的一些算法，可能带个buffer之类的

ByteXiaoTang, 2021-05-17:
[image]

此处，愿意是判空，结果is_debounce和its_retention值为默认2ms.防抖debounce用了起来，就是你看到的2ms

closed 2021-05-17 by yooouxin