COVESA/vsomeip #253: $ ./service-example   Configuration module could not be loaded!

Miaoishappening, 2021-05-19:


ByteXiaoTang, 2021-05-20:
配置文件和环境变量加载了么！

Miaoishappening, 2021-05-20:
> 配置文件和环境变量加载了么！

还要配置文件和环境变量嘛?这个10分钟的介绍里面也没有诶。。。
请问一下这个怎么配置呢?能不能详细一些。

ByteXiaoTang, 2021-05-20:
具体可参考readme

Miaoishappening, 2021-05-24:
> 具体可参考readme

谢谢问题已经解决了. :D

Miaoishappening, 2021-05-24:
Solution: 
You should build vsomeip like this:

```
$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/ ..
$ cmake -DENABLE_SIGNAL_HANDLING=1 .. //added line
$ make 
$ make install
```

closed 2021-05-24 by Miaoishappening

oska874, 2021-06-14:
> > 具体可参考readme
> 
> 谢谢问题已经解决了. :D

你是怎么解决的？我按照readme的流程，make&install，但是运行hello_world_service 还是会报相同的错误。

Miaoishappening, 2021-06-16:
> > > 具体可参考readme
> > 
> > 
> > 谢谢问题已经解决了. :D
> 
> 你是怎么解决的？我按照readme的流程，make&install，但是运行hello_world_service 还是会报相同的错误。

通过上面我的solution的步骤走还是不可以嘛？

Daniel-li121, 2021-06-18:
> > > > 具体可参考readme
> > > 
> > > 
> > > 谢谢问题已经解决了. :D
> > 
> > 
> > 你是怎么解决的？我按照readme的流程，make&install，但是运行hello_world_service 还是会报相同的错误。
> 
> 通过上面我的solution的步骤走还是不可以嘛？

after those steps you mentioned, refresh the cache configuration with command:
`sudo ldconfig`

SomnathHolkar, 2021-12-23:
I am also facing same issue.
I followed the exact steps mentioned as below.

$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/ ..
$ cmake -DENABLE_SIGNAL_HANDLING=1 .. //added line
$ make 
$ make install

eddiewangluminar, 2022-03-24:
@SomnathHolkar According to [Issue 139](https://github.com/COVESA/vsomeip/issues/139) it's because `libvsomeip3-cfg.so` couldn't be found. It can be solved by adding this prebuild library's path to `LD_LIBRARY_PATH`.

semajz, 2022-08-22:
“sudo ldconfig” worked~