COVESA/vsomeip #261: VSOMEIP disconnecting clients after service discover TTL on QNX service

kheaactua, 2021-06-30:
Any idea my vsomeip service disconnecting on QNX?

We ported vsomeip to QNX - I'm not sure I'm at liberty right now to share the exact patch, but in one of two implementations, we basically followed the WIN32 code flow, in the other the implementation is a little different, but both are exhibiting the same behaviour as described here.

The problem is that when the service is running on QNX with a client on another host, after the TTL defined in the service discovery section expires, the service disconnects.

For reference, every other configuration I've tried (QNX (client) ⟷ AOSP (service), QNX (client) ⟷ Linux (service), AOSP (client) ⟷ Linux (service), AOSP (service) ⟷ Linux (client), _etc_) all work, this behaviour is only exhibited when QNX is running the service.  This keeps open the possibility that our porting job to QNX isn't handling something properly, but I'm still not sure.

If I increase the TTL, the service runs longer - until that TTL expires.

I've been chasing this for a while.  I've had some issues with the debugger on QNX so I've resorted to print statements.

I'm not sure if I'm on to something, but there seems to be an issue with setting the event groups.  I've posted some output at the point it appears to fail (below).  Before this point everything is working as expected.

Note that this is my own CommonAPI vsomeip app, but I see the same with every vsomeip app I've tried, even `vsomeip/examples/hello_world`.

#### Client Behaviour

6 calls to `service_discovery_impl::create_eventgroup_entry`, two of which fail: The failures are the endpoints being reliable but the reliability type being unreliable.  That leads `create_eventgroup_entry` to return without creating the event group.
```
[warning] create_eventgroup_entry: Didn't insert subscription as subscription doesn't match reliability type: [0400.5678.80f2] 2
2021-05-30 08:15:32.937116 [info] update_routing_info: elapsed=1000 : delete service/instance 0400.5678
Hello_World service is NOT AVAILABLE
```

#### Server Behaviour

Starting from `check_stop_subscribe_subscribe`, the TTL value here is read as `0`.  Here's a "_traceback_" of sorts:
```
2021-05-30 08:15:07.077750 [info] Matt: check_stop_subscribe_subscribe:3250: ttl=0, iter->type != STOP_SUBSCRIBE_EVENTGROUP -> false
2021-05-30 08:15:07.077760 [info] Matt: check_stop_subscribe_subscribe:3269: returning false
2021-05-30 08:15:07.077760 [info] Matt: on_message:1149: calling process_eventgroupentry, ttl=0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077760 [info] Matt: process_eventgroupentry:1744: start
2021-05-30 08:15:07.077760 [info] Matt: process_eventgroupentry:1757: ttl=0
2021-05-30 08:15:07.077760 [info] Matt: process_eventgroupentry:2167: calling handle_eventgroup_subscription start: ttl = 0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077760 [info] Matt: handle_eventgroup_subscription:2207: start: ttl = 0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077760 [info] Matt: handle_eventgroup_subscription:2212: ndef VSOMEIP_ENABLE_COMPAT
2021-05-30 08:15:07.077770 [info] Matt: handle_eventgroup_subscription:2386: calling on_remote_unsubscribe: _ttl = 0
2021-05-30 08:15:07.077770 [info] Matt: on_remote_unsubscribe:2801: on_remote_unsubscribe
2021-05-30 08:15:07.077770 [info] Matt: check_stop_subscribe_subscribe:3250: ttl=0, iter->type != STOP_SUBSCRIBE_EVENTGROUP -> false
2021-05-30 08:15:07.077770 [info] Matt: check_stop_subscribe_subscribe:3269: returning false
2021-05-30 08:15:07.077770 [info] Matt: on_message:1149: calling process_eventgroupentry, ttl=0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077770 [info] Matt: process_eventgroupentry:1744: start
2021-05-30 08:15:07.077770 [info] Matt: process_eventgroupentry:1757: ttl=0
2021-05-30 08:15:07.077780 [info] Matt: process_eventgroupentry:2167: calling handle_eventgroup_subscription start: ttl = 0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077780 [info] Matt: handle_eventgroup_subscription:2207: start: ttl = 0, is_stop_subscribe_subscribe = 0
2021-05-30 08:15:07.077780 [info] Matt: handle_eventgroup_subscription:2212: ndef VSOMEIP_ENABLE_COMPAT
2021-05-30 08:15:07.077780 [info] Matt: handle_eventgroup_subscription:2386: calling on_remote_unsubscribe: _ttl = 0
2021-05-30 08:15:07.077780 [info] Matt: on_remote_unsubscribe:2801: on_remote_unsubscribe
selectiveTime client UNSUBSCRIBED
```

Again, this is after successful back and forth communication (RPC calls) for `TTL` seconds.

## Details:

QNX: `7.0.4`
VSOMEIP:
 - *QNX*:  based on `3.1.20.3`
 - *AOSP*: based on `3.1.20.3`
 - *Linux*: `3.1.20.3`

## Config Files

These are the one's I'm using now, but even with variations of these, or the one one `vsomeip/examples/hello_fnv` (modified to have service discovery) I get the same results

```
QNX_IP=192.168.2.54
LINUX_LAN_IP=192.168.2.53
SERVICE_DISCOVERY_PROTOCOL=udp
```

### Service config file:

`basic-service.qnx-client.linux.json.in`:
```json
{
  "unicast": "@QNX_IP@",
  "logging": {
    "level": "trace",
    "console": true,
    "version": {
      "enable": false
    }
  },
  "tracing": {
    "enable": true,
    "sd_enable": true
  },
  "applications": [
    {
      "name": "service-sample",
      "id": "0x0112"
    }
  ],
  "services": [
    {
      "service": "0x400",
      "instance": "0x5678",
      "reliable": {
        "port": "30510",
        "enable-magic-cookies": true
      },
      "is_reliable": true
    },
    {
      "service": "1025",
      "instance": "22236",
      "reliable": {
        "port": "30512",
        "enable-magic-cookies": true
      },
      "is_reliable": true
    }
  ],
  "supports_selective_broadcasts": [
    {
      "address": "@LINUX_LAN_IP@"
    },
    {
      "address": "@QNX_IP@"
    }
  ],
  "routing": "service-sample",
  "service-discovery": {
    "enable": true,
    "multicast": "224.224.224.245",
    "port": "30490",
    "protocol": "@SERVICE_DISCOVERY_PROTOCOL@",
    "initial_delay_min": "10",
    "initial_delay_max": "100",
    "repetitions_base_delay": "200",
    "repetitions_max": "3",
    "ttl": "4",
    "cyclic_offer_delay": "2000",
    "request_response_delay": "1500"
  }
}
```

### Client Config File

`basic-client.linux-service.qnx.json.in`:

```json
{
  "unicast": "@LINUX_LAN_IP@",
  "logging": {
    "level": "trace",
    "console": true,
    "version": {
      "enable": false
    }
  },
  "tracing": {
    "enable": true,
    "sd_enable": true
  },
  "applications": [
    {
      "name": "client-sample",
      "id": "0x0114"
    }
  ],
  "supports_selective_broadcasts": [
    {
      "address": "@QNX_IP@"
    },
    {
      "address": "@LINUX_LAN_IP@"
    }
  ],
  "routing": "client-sample",
  "service-discovery": {
    "enable": true,
    "multicast": "224.224.224.245",
    "port": "30490",
    "protocol": "@SERVICE_DISCOVERY_PROTOCOL@",
    "initial_delay_min": "10",
    "initial_delay_max": "100",
    "repetitions_base_delay": "200",
    "repetitions_max": "3",
    "ttl": "4",
    "cyclic_offer_delay": "2000",
    "request_response_delay": "1500"
  }
}
```

kheaactua, 2021-06-30:
Added successful feedback

lutzbichler, 2021-07-10:
Did you check the multicast offer messages in case of the service being run on QNX?

kheaactua, 2021-07-12:
It seems that with boost 1.63, the callback registered with:

```cpp
    offer_debounce_timer_.async_wait(
            std::bind(&service_discovery_impl::on_offer_debounce_timer_expired,
                      this, std::placeholders::_1));
```

[service_discovery_impl.cpp#L2674-L2676](https://github.com/GENIVI/vsomeip/blob/13f9c89ced6ffaeb1faf485152e27e1f40d234cd/implementation/service_discovery/src/service_discovery_impl.cpp#L2674-L2676)

that is suppose to get called after a certain amount of time decided by the timer *isn't* being triggered on QNX.

My colleague experimented with this code, and reported that he just couldn't get this callback to with in vsomeip with boost 1.63.

This problem appears to be resolved by upgrading to boost 1.74.  I'm still experimenting with it though.

closed 2021-07-16 by kheaactua

yunfeiyangcc, 2021-11-01:
@kheaactua Is boost 1.74 ok on QNX? I have a same problem.  [https://github.com/COVESA/vsomeip/issues/261#issuecomment-878187234](url)

kheaactua, 2021-11-01:
hey @yunfeiyangcc , yep, boost 1.74 appears to have fixed that problem on QNX