COVESA/vsomeip #263: Configuration option deserialize bug.

yunfeiyangcc, 2021-07-07:
@lutzbichler 

Hi lutzbichler,

There are two IPv4 option and a configuration option in the subscribe message.
Order as follows:
0: Configuration Option
1: IPv4 Endpoint Option (192.168.116.131:41234(TCP))
2: IPv4 Endpoint Option (192.168.116.131:44234(UDP))
[image]

When the SD receives this subscribe message and deserialize it.
Because the remaining_length is set by options_length_(45), the length of configuration option is 18. 
**so it cannot stop deserializing until the end of the subscribe message.**

I made a change in function(configuration_option_impl::deserialize), please check it.
[image]

lutzbichler, 2021-07-10:
I am afraid I do need to see the message bytes to understand the root cause.

ims0, 2021-12-08:
I also encountered this problem. The solution is to exit if the length is equal to 0.
[image]

DiogoPedrozza, 2023-06-05:
Hi all.
Fix also introduced in [3.3.5.1](https://github.com/COVESA/vsomeip/releases/tag/3.3.5.1) and present on all the subsequent releases.
Closing the issue.

closed 2023-06-05 by DiogoPedrozza