COVESA/vsomeip #264: Netlink send error : Invalid argument

hmluqman, 2021-07-07:
I am trying to build and run vsome IP basic examples on Ubuntu 20.05 in Widnows 10 as WSL. 
I follow the procedure describe in[ vsomeip in 10 minutes](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes). I am following Availability section where one service is offered and client notifies if available. 

The output at service side is 

`2021-06-07 18:08:37.073496 [warning] Netlink send error : Invalid argument
2021-06-07 18:08:37.079674 [info] Network interface "n/a" state changed: up
2021-06-07 18:08:37.085017 [info] Route "n/a" state changed: up
2021-06-07 18:08:37.091796 [info] udp_server_endpoint_impl: SO_RCVBUF is: 163840
2021-06-07 18:08:37.102113 [info] udp_server_endpoint_impl: SO_RCVBUF (Multicast) is: 163840
2021-06-07 18:08:37.109820 [error] udp_server_endpoint_impl::join:set_option: Invalid argument
2021-06-07 18:08:37.116354 [info] Port configuration missing for [1234.5678]. Service is internal.
2021-06-07 18:08:37.121588 [info] SOME/IP routing ready.`

It does not able to detect network interface which should have been loopback, 'lo'. It is detecting as 'n/a'. Client does not know any service. 

How can I solve this issue?

lutzbichler, 2021-07-10:
The network confguration as well as the vsomeip configuration are needed to help here. However, does WSL support netlink at all?

fredani, 2021-10-13:
Thanks! Finally, I got it up n running!

closed 2023-03-03 by goncaloalmeida