COVESA/vsomeip #265: Routing manager crashes due to failed TCP socket bind

DrewDAGator, 2021-07-13:
If the routing manager fails to bind to the specific port that it allocates for the local server endpoint, the vsomeip application will crash due to an attempt to dereference a null pointer. This can be caused by the port being blocked or it being already in use by another socket. The try/catch block in `endpoint_manager_base::create_local_server` will catch the exception, but the function will return a null pointer, which causes the next line to dereference a null pointer.

Here's the code where it crashes in this scenario:

```
    local_receiver_ = std::static_pointer_cast<endpoint_manager_base>(
            host_->get_endpoint_manager())->create_local_server(shared_from_this());
    local_receiver_->start();
```

Link: https://github.com/GENIVI/vsomeip/blob/master/implementation/routing/src/routing_manager_stub.cpp#L1626

The suggested fix would be to either:
1. Pick a random available port instead of binding to a designated port (if possible)
2. Check whether `local_receiver_` is null prior to calling `local_receiver_->start()`

ByteXiaoTang, 2021-07-17:
the port should not be binded by other process in the soc. so you can use port range or just use the port other process not use

DrewDAGator, 2021-07-19:
Yeah the workaround is to find a client id that doesn't result in it failing to bind. The TCP port it tries to bind to is `client_id + 51234`

lutzbichler, 2021-08-04:
You can adapt VSOMEIP_INTERNAL_BASE_PORT in internal.hpp.in to use a free range. The current implementation (on Windows) cannot use a random port as the clients would not find each other anymore.

closed 2021-08-04 by lutzbichler